microsoft/Qcodes #767: Change parameters to be type-specific

nulinspiratie, 2017-09-28:
I've been using parameters a lot as object attributes, both for instrument drivers, and also for objects that are completely unrelated to physical instruments (meta-instruments).
Doing so has some clear advantages:
- Integration with QCoDeS loops (makes them sweepable and measurable),
- Allows a standard way to add attribute metadata (e.g. docstring, labels, validators, etc.),
- Integration with extra tools, such as a monitor.

However, one of the things that I've struggled with is when exactly to use a parameter, and when to use straightforward object attributes (`self.attr = some_value`). This becomes especially relevant for meta-instruments, as there not every parameter corresponds to an instrument instruction. The root of the problem seems to be that parameters don't really behave the same way as attributes. For example, getting/setting a parameter is done by calling it (`param(new_val)`), instead of simply assigning a value. (`param(new_val)` instead of `param = new_val`). This leads to the annoying situation where I have to think for every attribute if it actually is a parameter or not. 

Ideally, we would want parameters to behave more like attributes, in which case we wouldn't have to worry about if an attribute is actually a parameter or not. any attribute relevant for QCoDeS could be a parameter (or even all attributes, although this might cause unnecessary overhead). I've been thinking about a way to tackle this problem, and my proposed solution is to have type-specific descriptor parameters. As an example, say we have a ListParameter:

```python
class ListParameter(list, Parameter)
    def __get__(self, instance, owner):
        return self
    def __set__(self, instance, val):
        self(val)

list_parameter = ListParameter()
```

By having the ListParameter inherit from both `list` and `Parameter`, we allow it to behave as a list, and get the additional functionality of a parameter. We can therefore treat it exactly like a list (e.g. `list_parameter.append(val)`, while it also has all the Parameter features (e.g. `list_parameter.unit`).

Furthermore, if we attach the ListParameter to an instrument (`instr.list_parameter`), the descriptor methods (`__get__` and `__set__`) allow simple getting/setting of values (e.g. `instr_list_parameter = new_val`). Note that this only works when the ListParameter is an attribute of another object, otherwise we would still need to use `list_parameter(new_val)`. This could also solve a few other issues. A MultiParameter would simply be a ListParameter, validators would usually be implicit, but could still be overridden for additional conditions such as being a multiple of a value. 

One of the main issues with these type-specific parameters is dealing with immutable types, such as `int`. In this case, we cannot change its value once it's instantiated, and so we need another solution for this. The only solution I have come up with so far is to have the IntParameter mimick the int class, but being mutable. However, I can imagine there could be a better solution, so if someone has one please come forward :)

nulinspiratie, 2017-09-28:
I realized that descriptors have to be defined for a class, not an instance, and so the `instr.list_parameter = new_val` functionality might have to be implemented in the parent class `__setattr__` instead.

nulinspiratie, 2017-10-12:
Could anyone give their two cents on this? I was hoping to work on this soon, as we need this for our next goal, a monitor-like IPython widget for Jupyter notebook

@jenshnielsen @WilliamHPNielsen @sohailc @AdriaanRol

WilliamHPNielsen, 2017-10-12:
Hi @nulinspiratie, sorry for being slow in reponding to this.

As far as I can see, the fact (which you mention) that descriptors can only be instantiated at class level is a big deal breaker. Can you perhaps flesh out how to circumvent this inconvenience? It would be a pain not to have two instances of being able to differ in which parameters they have.  It seems that we would then have to subclass rather than instantiate (not cool!).

As an example, consider the `Keysight_34465A` driver. There, a keyword argument `DIG` to the `__init__` controls whether some extra parameters (available only on instruments with the purchase of the `DIG` upgrade) are added to the instrument. How would we do this using descriptor parameters?

WilliamHPNielsen, 2017-10-12:
Another small comment: I like the idea of parameters (and everything else!) having a type. Note though, that validation rarely stops at the type; usually it checks for values out of some range. This range will in general depend on the value of other parameters (like a function generator whose max output amplitude depends on the signal frequency), and this is another currently unsolved problem in QCoDeS.

AdriaanRol, 2017-10-12:
@nulinspiratie I'm not a fan of making these changes. I'm not a fan for the following reasons 
- I think parameters are work very well in their current form. 
- There is an advantage to how explicit parameters are in defining get and set. To me it is very clear what a parameter is and how it works. I'm afraid this clarity get's lost if many different parameter classes are introduced, especially when mixing them with other data type classes. 
- I'm afraid of backwards compatibility. We have a massive legacy codebase. 

Basically my concerns come down to a question of motivation. I'm not really convinced this is the best solution to the problem as it seems rather radical. This may be just my lack of understanding of the idea through. 

When writing this I have to say I do feel the problem of `instr.attr = val ` vs `instr.par(val)`. I vaguely remember that @alexcjohnson had a very good reason to want to always explicitly set a parameter instead of setting it using the equal sign.

WilliamHPNielsen, 2017-10-12:
@AdriaanRol could Alex' very good reason have been rather in getting than setting? There is a crucial difference between `get` and `get_latest`.

jenshnielsen, 2017-10-12:
@AdriaanRol I think if we want to go this was we would have to implement a new parameter class and then slowly convert instruments with a very long deprecation cycle (perhaps indefinite) There is IMHO no good way to introduce this in the existing Parameter without major breakage

alexcjohnson, 2017-10-12:
Thanks for the ping @AdriaanRol 

Early on we had some discussion of using `@property` for parameters, and my primary argument against this is that it makes it difficult to pass a parameter around as a first-class object - as `@property` you would only be able to grab the parameter's *value*, not its *object* (or only naturally able, I'm sure we could make some syntax for it but it would not be the simple `instrument.parameter`). From that starting point, using `instrument.parameter` as a getter is ruled out. I suppose one could still define the setter to work like a `@property`, ie `instrument.parameter = value` (and that would be a way I guess to ensure users don't accidentally overwrite their parameter objects) but it feels confusing to me to mix those behaviors, and also wouldn't be able to still work if you pull the parameter out of the instrument: `instrument.parameter = 5` might work, but `p = instrument.parameter; p = 5` would not (perhaps you can hack python to make this work, but it would cause other confusion - how would you assign `p` to represent a different parameter?).

@nulinspiratie my glib answer to this would be to make everything a parameter, no raw attributes! Even if it doesn't correspond to a physical state variable, each value you are storing as an attribute still has some effect on the experiment you're doing, so having it go through the same machinery (for logging, for validation, for connecting to GUI tools) would have lots of benefits.

AdriaanRol, 2017-10-12:
@alexcjohnson Thanks for clarifying that .
>it makes it difficult to pass a parameter around as a first-class object - as @property you would only be able to grab the parameter's value, not its object 

We  use that functionality a lot. 

>@nulinspiratie my glib answer to this would be to make everything a parameter, no raw attributes! 

This is also how I use parameters (and how I try to teach everyone).

nulinspiratie, 2017-10-13:
Hey guys thanks for the useful input! I understand it would be quite a radical change for the parameters, and I must admit that I don't have a completely worked out idea. I just wanted to put this idea forth to see if it's a viable option or not.
While I agree with @adriaanrol that parameters work well in their current form, especially for drivers, I'm running into its limitations with more abstract classes unrelated to drivers.

To give a concrete example, we use pulse sequences that consists of `Pulse` objects. Each `Pulse` has attributes `t_start`, `t_stop`, and `duration`. Now `pulse.t_start` and `pulse.duration` are things we could want to vary in a measurement, so we want them to be parameters. `pulse.t_stop(=pulse.t_start + pulse.duration)` on the other hand is only there for convenience and is much easier to implement as a dependent property (`@property`). If we now want the center time of the pulse, it would be `(pulse.t_start() + pulse.t_stop) / 2`, i.e. we would have to remember that t_start is a parameter and t_stop is not. Of course we could also do `pulse.t_start() + pulse.duration()/2`, so this example has a loophole, but I have been runnning into similar situations quite often. The main point I'm trying to make here is that forcing every attribute of every object to be a parameter places quite a large contraint on objects, especially since many such attributes may never ever need the functionality that a parameter provides. Some classes don't need any parameters, but do interface with other classes that do, would these then also need all its attributes to be parameters, or do we need to remember which classes contain parameters and which ones do not? my sense is that having two types of attributes (standard attributes and parameters) that behave in different ways will start to cause issues the larger the framework around QCoDeS becomes.

The main proposal is therefore to change the parameter from being a container for a value to being (an extension to) the value itself. That we would need type-specific parameters is merely a consequence of this. From the outside, these parameter attributes behave just as the type it represents. @alexcjohnson this also solves one of the issue you raised, as you would simultaneously pass both the value and the parameter.

To be clear, the parameters would still have get/set functionality that can be called via brackets. Coincidentally, this should solve most of the major backwards compatibility issues, as parameters can still operate just as before. 
@WilliamHPNielsen My idea is to step away from descriptors, and instead use instrument.__setattr__ to implement `instrument.parameter = value`. This way, we can still define parameters for each class instance. @alexcjohnson you are right that `p = instrument.parameter; p = 5` would not work, but you can still do `p(5)`, and it should make sense that `p = 5` wouldn't update the parameter value, but create an integer p with value 5.

However, one important issue raised by @WilliamHPNielsen that I hadn't thought much about before is the difference between get and get_latest. Should calling `instrument.parameter` update its value by callling `instrument.parameter.get()`, or simply keep its latest value? The former might lead to issues that it calls a `.get` when it shouldn't, while the latter can lead to situations where `instrument.frequency` shows an old value that is not up to date. I'm not sure what the best solution is. I veer towards the latter, which is also easier to implement. In this case, a `.get` is only called during a Loop, and with `instrument.parameter()`. But perhaps there's a better solution I haven't thought of.

@WilliamHPNielsen Type-specific parameter still need validators to ensure that an IntParameter will have an integer value. These validators can still have additional constraints (e.g. multiples of 2)

I'm hoping this could serve as a bridge between having many attributes as parameters, while not having to worry if an attribute is a parameter or not. Parameters could still be used in the old way (except changing type). I understand the point brought up by @adriaanrol that too many parameter type might complicate things, though if their behavior is identical except for their type properties this should be manageable.The major issue I'm seeing is how to deal with get versus get_latest, would this be a deal breaker?

nulinspiratie, 2017-10-24:
Hmm I realized that it's probably not a good idea to make non-mutable types such as ints mutable, as it can cause unintended issues. I'll close this, though I'm still hoping there could be a way to make parameters behave more pythonic...

closed 2017-10-24 by nulinspiratie