microsoft/Qcodes #768: Optionally raise an exception if a Manual Parameter returns None /validate on get

AdriaanRol, 2017-09-28:
We use a lot of ManualParameters in our meta instruments (I guess @nulinspiratie you do something similar). Sometimes we forget to set the value of such a parameter. When we then try to use it we get strange type errors resulting from using the `None` as a value somewhere else. 

Below is an example of such an exception that can be hard to debug if you are not aware of this. 
```
d:\githubrepos\pycqed_py3\pycqed\instrument_drivers\meta_instrument\qubit_objects\CCL_Transmon.py in _prep_td_sources(self)
    646         # Set source to fs =f-f_mod such that pulses appear at f = fs+f_mod
    647         self.instr_LO_mw.get_instr().frequency.set(
--> 648             self.freq_qubit.get() - self.mw_freq_mod.get())
    649 
    650         self.instr_LO_mw.get_instr().power.set(self.mw_pow_td_source.get())

TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
```

I would propose adding a validate on get (`Boolean`) attribute to the parameter class so that we get the full context power (and checks) of the validators.

AdriaanRol, 2017-09-28:
@nulinspiratie maybe this is already part of your recent parameter improvements?

nulinspiratie, 2017-09-29:
@AdriaanRol It isn't at the moment. I think the question is how we want the parameter to behave before it has a value. Would the correct behaviour be to return None, or to raise an error immediately. I'm undecided on the matter, as it is sometimes useful to have parameters that can be initialized/set to None. It might also introduce bugs elsewhere if a parameter can raise errors during a get