microsoft/Qcodes #769: Recent change to PyQTgraph plotting API breaks plotting with units

AdriaanRol, 2017-09-28:
I just merged QCoDeS master into our fork of QCoDeS and found that it breaks our live plotting. 
I'll try to go to 0.1.7 now so that I have recent changes that I require. 

Digging a bit deeper it seems that this is related to a change introduced by 35400a60448101c66337d9b3373b9af1c9666725

I have attached the relevant part of my traceback pointing to the problem. 
I'll also add the relevant lines of codes with a bit more context below. 

```
d:\githubrepos\pycqed_py3\pycqed\measurement\measurement_control.py in measure(self, *kw)
    171     def measure(self, *kw):
    172         if self.live_plot_enabled():
--> 173             self.initialize_plot_monitor()
    174 
    175         for sweep_function in self.sweep_functions:

d:\githubrepos\pycqed_py3\pycqed\measurement\measurement_control.py in initialize_plot_monitor(self)
    570                                      subplot=j+1,
    571                                      color=color_cycle[j % len(color_cycle)],
--> 572                                      symbol='o', symbolSize=5)
    573                 self.curves.append(self.main_QtPlot.traces[-1])
    574                 j += 1

d:\githubrepos\qcodes\qcodes\plots\base.py in add(self, updater, *args, **kwargs)
     89         # TODO(giulioungaretti): replace with an explicit version, see expand trace
     90         self.expand_trace(args, kwargs)
---> 91         self.add_to_plot(**kwargs)
     92         self.add_updater(updater, kwargs)
     93 

d:\githubrepos\qcodes\qcodes\plots\pyqtgraph.py in add_to_plot(self, subplot, **kwargs)
    147         if prev_default_title == self.win.windowTitle():
    148             self.win.setWindowTitle(self.get_default_title())
--> 149         self.fixUnitScaling()
    150 
    151     def _draw_plot(self, subplot_object, y, x=None, color=None, width=None,

d:\githubrepos\qcodes\qcodes\plots\pyqtgraph.py in fixUnitScaling(self, startranges)
    528             for axis in ('x', 'y', 'z'):
    529                 if self.traces[i]['config'].get(axis):
--> 530                     unit = self.traces[i]['config'][axis].unit
    531                     if unit not in standardunits:
    532                         if axis in ('x', 'y'):

AttributeError: 'list' object has no attribute 'unit'
```

This is the method that hits the offending line when adding a plot. 
```python 
    def initialize_plot_monitor(self):
        # new code
        if self.main_QtPlot.traces != []:
            self.main_QtPlot.clear()
        self.curves = []
        xlabels = self.sweep_par_names
        xunits = self.sweep_par_units
        ylabels = self.detector_function.value_names
        yunits = self.detector_function.value_units

        j = 0
        if (self._persist_ylabs == ylabels and
                self._persist_xlabs == xlabels) and self.persist_mode():
            persist = True
        else:
            persist = False
        for yi, ylab in enumerate(ylabels):
            for xi, xlab in enumerate(xlabels):
                if persist:  # plotting persist first so new data on top
                    yp = self._persist_dat[
                        :, yi+len(self.sweep_function_names)]
                    xp = self._persist_dat[:, xi]
                    if len(xp) < self.plotting_max_pts():
                        self.main_QtPlot.add(x=xp, y=yp,
                                             subplot=j+1,
                                             color=0.75,  # a grayscale value
                                             symbol='o', symbolSize=5)
                self.main_QtPlot.add(x=[0], y=[0],
                                     xlabel=xlab,
                                     xunit=xunits[xi],
                                     ylabel=ylab,
                                     yunit=yunits[yi],
                                     subplot=j+1,
                                     color=color_cycle[j % len(color_cycle)],
                                     symbol='o', symbolSize=5)
                self.curves.append(self.main_QtPlot.traces[-1])
                j += 1
            self.main_QtPlot.win.nextRow()
```

For reference self.main_QtPlot is below. `from qcodes.plots.pyqtgraph import QtPlot`
```python 
            self.main_QtPlot = QtPlot(
                window_title='Main plotmon of {}'.format(self.name),
                figsize=(600, 400))
```

jenshnielsen, 2017-09-28:
It's hard for us to have visibility on how you are using qcodes in your fork but in this case I guess we should just replace that line with a get and skip the rescaling if no unit is set

AdriaanRol, 2017-09-28:
Hi @jenshnielsen it may be a bit hard to see but the error is actually raised on the line where I am most definitely setting units (block slightly lower in the snippet I pasted). So I'm not sure what causes this. AFAIK I'm using the default API of the Qt_Plot object so I would expect this to work but it broke somehow. If your proposed solution solves the problem I'd be quite happy with it, I think not rescaling if it is not SI would make a lot of sense (and the other changes that were introduced). 

Also I just merged 0.1.7 and the speed increase (#712 ) when setting up a plot is very noticable, I would not have found that! :+1: .

jenshnielsen, 2017-09-28:
I think #770 will do the trick can you test that

jenshnielsen, 2017-09-28:
Is your version online somewhere so I can test it. The issue behind #712 is really odd I have not had the time to dive into pyqt/qt to understand what happens

AdriaanRol, 2017-09-28:
@jenshnielsen Let me try #770 . Our version is online at https://github.com/DiCarloLab-Delft/Qcodes. 
I'll be going to a talk in 5 mins but let me try before that. 

Also I think this behaviour is testable quite easy. I have Travis compatible tests set up for a lot of things involving the live plotting, if you want I can show you how we do it.

astafan8, 2018-06-14:
So far as I understand from the discussion, PR #805 fixed this, and it's merged, so I am closing this issue.

closed 2018-06-14 by astafan8