microsoft/Qcodes #775: Bug: Saving metadata issue when dict contains both numeric and non-numeric keys

nulinspiratie, 2017-10-03:
### Steps to reproduce
1. Create: `p = Parameter(name='test', val_mapping={1: 1, '2': 2})`
2. Perform measurement with p

### Expected behaviour
Measurement should run without problems

### Actual behaviour
Saving the metadata as a JSON fails because it tries to sort the keys.
Problem lies in lines 338-339 of qcodes/data/gnuplot_format.py:

```
json.dump(data_set.metadata, snap_file, sort_keys=True,
                 indent=4, ensure_ascii=False, cls=NumpyJSONEncoder)
```

### Proposed solution
Remove sort_keys, is it really necessary to sort them? Alternatively, force all dict keys to be of same type. Some instruments (e.g. SR830) will need modifications

@jenshnielsen @WilliamHPNielsen @sohailc

jenshnielsen, 2017-10-03:
I could be talked in to both solutions. 

On one hand I don't think the sorting is particularly important 
on the other hand all keys will be serialised to strings anyway when dumping to JSON so perhaps it makes sense to standardise? 

It seems to me that that is a bug in the python JSON library that sorting happens before conversion to string but its the case in both the python 3.5 and 3.6 releases that I tested so I guess we have to live with that.

nulinspiratie, 2017-10-25:
@jenshnielsen I guess pre-converting to strings would also work, though this can fail if the string representation would have a different sorting order, e.g. `'2' > '1e9'`

peendebak, 2017-10-26:
@jenshnielsen @nulinspiratie The same problem arises with the SR830 driver. A minimal example constructed from the driver:
```
import qcodes

class Dummy(qcodes.Instrument):
    
    def __init__(self, name, **kwargs):
        super().__init__(name,**kwargs)
        
        self.add_parameter('p', set_cmd=None)
        self.add_parameter('q', set_cmd=None)
        self.add_parameter('buffer_SR',
                           label='Buffer sample rate',
                           set_cmd=None,
                           initial_value=0,
                           unit='Hz',
                           val_mapping={0:0, 62.5e-3: 0,
                                        0.125: 1,
                                        0.250: 2,
                                        0.5: 3,
                                        1: 4, 2: 5,
                                        4: 6, 8: 7,
                                        16: 8, 32: 9,
                                        64: 10, 128: 11,
                                        256: 12, 512: 13,
                                        'Trigger': 14},
                           get_parser=int
                           )
        
d=Dummy('test6')
station=qcodes.Station(d)
ds=qcodes.Loop(d.p[0:10:1]).each(d.q ).run()
```

peendebak, 2017-10-26:
@nulinspiratie @jenshnielsen The code works if I use `import simplejson as json`. Would this be a workaround?

nulinspiratie, 2017-11-08:
@peendebak I guess that would be a solution, though it's not clear to me
why this is fixed in simplejson. Perhaps it was a recent bugfix?

peendebak, 2017-11-10:
Not sure why simplejson did work

@jenshnielsen @WilliamHPNielsen @nulinspiratie We have multiple users who cannot use recent versions of qcodes together with the SR830 driver. Do you have any idea when you can look into this issue?

WilliamHPNielsen, 2017-11-13:
I think we can live without the sorting. I'll make a PR soon (:P) to remove the sorting.

PR #868 (2017-11-14): Fix/nognuplotsorting
Fixes #775.

Changes proposed in this pull request:
- Make the `json.dump` of metadata not sort the keys
- Add a little test that this works

@jenshnielsen @nulinspiratie @peendebak
closed 2017-11-14 by WilliamHPNielsen