microsoft/Qcodes #777: driver/Keysight 33210 AWG

Dominik-Vogel, 2017-10-03:
Need for a driver for the Keysight 33210 AWG:

http://www.keysight.com/en/pd-1407410-pn-33210A/function-arbitrary-waveform-generator-10-mhz?pm=PL&nid=-536902257.793368&cc=DK&lc=dan

please refer to the Keysight 33250 AWG which is probably very similar.

https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/Keysight/Keysight_33500B_channels.py

WilliamHPNielsen, 2017-10-04:
We have that instrument in stock. I'll see if I can make a combined driver for all the Keysight/Agilent Function/ Arbitrary Waveform generators.

WilliamHPNielsen, 2017-10-04:
But for now, we can probably be satisfied with the one we already have:
https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/agilent/HP33210A.py

closed 2017-10-04 by WilliamHPNielsen