microsoft/Qcodes #840: Driver for the SR860

sohailc, 2017-11-02:
A driver for the SR860 is needed.

sohailc, 2017-11-02:
As I understand the SR650 driver also works for the SR860. If so, we can close this issue

WilliamHPNielsen, 2017-11-02:
How about the SR830? I would expect the two lock-ins to be very similar.

WilliamHPNielsen, 2017-11-02:
But there are probably some subtle differences, at least in valid parameter ranges. Perhaps we should make a common base class for the two intruments, or even just one class that they both use (but this shouldn't be called SR830 in that case)?

sohailc, 2017-11-20:
I tried to use the SR830 driver with the SR860 instrument and used the example notebook in the docs folder. The command "sr.ch1_display('X')" seems to fail with a timeout error so the SR830 driver is note usable with the SR860/SR865

sohailc, 2017-11-20:
The SR865 driver does as first glance work with the SR860 instrument. However, I need buffered acquisition to work with the SR860, which is why I will work on commands to make this work.

WilliamHPNielsen, 2017-11-21:
@sohailc I think it is worth spending some time uniting the two (soon to be three) drivers. I have only glanced at the two existing drivers just now, so it might be harder than I think, but it _seems_ very doable. Consider that parameters like `input_shield`, `frequency`, `input_coupling` and many more are identical in the two drivers. The differences could be very few. I think that writing a common base class, `SR_8XX` (or whatever) , that both lock-ins inherit from is worth the extra effort.

spauka, 2017-12-02:
Agreed that they should inherit from the same base. 

I will point out the caveat that the methods for pulling data `OUTP? x` have different values of x for pulling X,Y,R,theta compared to the SR830, and the commands for a number of operations (`SCAL` vs `SENS` for setting sensitivity for example) are different.

sohailc, 2017-12-08:
I think the SR830 and the SR860/SR865 are similar in a lot of respects, but there are also quite a few differences. A lot of the ranges are different, but there are other differences as well. The way that buffered readout works is different. I think there could be some shared code, but I think its more trouble then worth