microsoft/Qcodes #856: RTO1024/RTO1000 scope driver issues

AdriaanRol, 2017-11-07:
I have found several problems with the RTO scope driver. 

1. The when using the HD mode on an RTO1024 the bandwidht get's set to 1MHz and cannot be changed from there. 
This would not be problematic if this mode was not needed but it is also used to set the dataformat which uses the same parameter. 

Observe the following difference and not the (spurious) staircase due to digitization in the left figure. 

[image]

Both are the same trace measured. The one on the right is measured by a little hack in the driver. 
The second trace contains a scaling error on the y-axis but I'm sure that can be fixed quite easilty. 
[image]

I added a screenshot from the scope itself as reference. 
[image]

2. Recent QCoDeS updates make this driver give a warning upon instantiation. IMO this should be fixed. 
[image]

AdriaanRol, 2017-11-07:
When using this workaround it is important to also use the following change for the rescaling of the data. 

[image]

WilliamHPNielsen, 2017-11-08:
Hi @AdriaanRol, thanks for catching this one. I don't have access to an instrument right now, but let's see if I can make sense of what is going on.

From how I understood the manual, one can not acquire in 16 bit unless the instrument is in high definition mode. Is you left figure really 16 bit with the high definition mode off? (nevermind the data _format_, are there really 2^16 unique values?) If so, then we might remove the `high_definition_state` paramater. If not, I think a real solution is to make the high definition mode not limit the bandwidth. Could you try using the `HDEFinition:BWIDth` command to see if you can set the high definition mode bandwidth. 

I am sorry that I did not realise that the high definition mode comes with a bandwidth of its own.

AdriaanRol, 2017-12-06:
I just figured out how to increase the bandwidth when using HD mode. 

I think this should also be implemented when addressing this issue. 

Tested using : 
[image]
Where rto is an instance of the RTO instrument driver. 

Below is an excerpt of the manual (page 1059) 
[image]

WilliamHPNielsen, 2017-12-06:
Yes, that's what I tried to hint at above. Glad that we agree 😄 

Do you want to make a PR adding this parameter? I can also do it, but it will not happen this week.

WilliamHPNielsen, 2017-12-07:
Or actually, an RTO 1044 just landed on my desk for an entirely different reason. Let me seize the opportunity to fix this.

AdriaanRol, 2017-12-07:
Awesome, looking forward to the PR!

PR #907 (2017-12-14): driver/ RTO-1000 HDEF bandwidth issue
Fixes #856

Changes proposed in this pull request:
- Add a parameter controlling the high def mode bandwidth
- Document the use of this in the example notebook
- Add a simulated instrument and an initialisation test

@jenshnielsen @AdriaanRol
closed 2017-12-14 by WilliamHPNielsen