microsoft/Qcodes #902: Custom parameters break datasaver (new dataset)

dmtvanzanten, 2017-12-05:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.
```
pP = ParamSpec('P', 'number', label='Power', unit='W')

meas = Measurement()
meas.registerParameter(dac.ch1)  # register the first independent parameter
meas.registerParameter(pP)
meas.registerParameter(dmm.v1, setpoints=(dac.ch1,))  # now register the dependent oone

R = 10E6 // Voltage to current conversion
with meas.run() as datasaver:         
    for set_v in np.linspace(0, 25, 251):
        dac.ch1.set(set_v)
        get_v = dmm.v1.get()
        datasaver.addResult((dac.ch1, set_v),
                            (dmm.v1, get_v), ('P', set_v*get_v/R))
        
    dataid = datasaver.id  # convenient to have for plotting
```

### Expected behaviour
Data being saved including the calculated virtual parameter P.

### Actual behaviour
```
WARNING:qcodes.dataset.measurements:Could not commit to database; table name-1-27 has no column named P
Starting experimental run with id: 27
```

### System
**operating system**

**qcodes branch**
feature/dataset

**qcodes commit**

WilliamHPNielsen, 2017-12-05:
This was never supposed to work that way.

jenshnielsen, 2018-07-20:
@WilliamHPNielsen This should probably raise then? There is a registerCustomParameter now right

closed 2018-07-20 by jenshnielsen