microsoft/Qcodes #908: fix AMI430_3D driver: Parameters should not return arrays

sohailc, 2017-12-08:
The AMI430_3D driver has a number of parameters which return arrays, e.g.: 

"cartesian_measured", "cartesian", "spherical_measured", etc. 

As these are regular parameters instead of ArrayParameters, this is wrong. 

The proper fix is to make these parameters, ArrayParameters.

sohailc, 2017-12-08:
ArrayParameter does not have a set attribute, which means we cannot simply use the ArrayParameter class. There are three solutions: 

1) Use the normal Parameter class but store the value as a single string. 
2) Don't use parameters at all for saving/accessing Cartesian/spherical/cylindrical set point but use class methods. The down side of that is that we need separate get/set functions for  cartesian/spherical/cylindrical. 
3) Add a set method on the ArrayParameter class. This is by far the best way, but seeing that this has not been done yet, makes me suspect that this isn't trivial.

WilliamHPNielsen, 2017-12-11:
@sohailc As discussed on slack, you will go with solution 3., right?

sohailc, 2017-12-11:
Yes, I will go with solution 3