microsoft/Qcodes #922: warning when storing 2D data

peendebak, 2017-12-15:
PR #899 breaks writing a 2D dataset with setpoints arrays. We create a 2D dataset manually. The format is
```
DataSet:
   location = '2017-12-15/22-45-20_qtt_scan2D'
   <Type>   | <array_id>          | <array.name>        | <array.shape>
   Measured | keithley1_amplitude | keithley1_amplitude | (80, 50)
   Setpoint | VP2                 | VP2                 | (80,)
   Setpoint | VP1                 | VP1                 | (80, 50)
```
All values of `.last_saved_idx` are None. We then write to this dataset using a double for loop:
```
    for ix, x in enumerate(stepvalues):
        alldata.store((ix,), {'VP2': x})
        
        for iy, y in enumerate(sweepvalues):
            datapoint ={'VP1': y, 'keithley1_amplitude': x+y}            
            alldata.store((ix,iy), datapoint)
```
This generates a warning `format.py` here: https://github.com/QCoDeS/Qcodes/blob/master/qcodes/data/format.py#L220

After the first ` alldata.store((ix,), {'VP2': x})` with `ix=0` we have `alldata.VP1.last_saved_index` equal to `3999`. How can we prevent this warning? Are can we change the warning in a specific type we we can ignore it?

@jenshnielsen

jenshnielsen, 2017-12-16:
@peendebak Does it break or just generate a warning?

jenshnielsen, 2017-12-16:
But this warning exactly means that after the first call to store the data set thinks that up to index 3999 has been written to file but the file does not in fact exist. It seems to me that is a hint that something is wrong and we should handle that

peendebak, 2017-12-18:
@jenshnielsen It only generates a warning. The example code that I posted seems like a valid way to use the dataset, so I think the warning should be removed or the formatter improved.

jenshnielsen, 2017-12-18:
@peendebak Could you provide a complete example of what you are doing because I can't reproduce the issue with the code below 

```python
import qcodes as qc
loc_provider = qc.data.location.FormatLocation(fmt='data/{date}/#{counter}_{name}_{time}')
qc.data.data_set.DataSet.location_provider=loc_provider

outer_size = 300
inner_size = 1000

vp2 = qc.DataArray(name='VP2', full_name='VP2', label='VP2', array_id='VP2', shape=(), is_setpoint=True)
vp2.nest(size=outer_size)
vp2.init_data()
vp1= qc.DataArray(name='VP1', full_name='VP1', label='VP1', array_id='VP1', shape=(), is_setpoint=True)
vp1.nest(size=inner_size, set_array=vp1)
vp1.nest(size=outer_size, set_array=vp2)
vp1.init_data()
ka = qc.DataArray(name='keithley1_amplitude', full_name='keithley1_amplitude',
                  label='keithley1_amplitude', array_id='keithley1_amplitude',
                  shape=(), is_setpoint=False)
ka.nest(size=inner_size, set_array=vp1)
ka.nest(size=outer_size, set_array=vp2)

ka.init_data()
alldata = qc.new_data()
alldata.add_array(vp2)
alldata.add_array(vp1)
alldata.add_array(ka)

alldata

xvals = range(outer_size)
sweepvalues = range(inner_size)
for ix, x in enumerate(xvals):
    alldata.store((ix,), {'VP2': x})
    for iy, y in enumerate(sweepvalues):
        datapoint ={'VP1': y, 'keithley1_amplitude': x+y}            
        alldata.store((ix,iy), datapoint)
alldata.finalize()
```

jenshnielsen, 2018-02-13:
@peendebak can you still reproduce this with the correct init code?

jenshnielsen, 2018-03-13:
Closing do to lack of response. It seems most likely that the dataset was used incorrectly but without a minimal example there is no way to tell

closed 2018-03-13 by jenshnielsen