microsoft/Qcodes #926: Drop support for python 3.5

WilliamHPNielsen, 2017-12-18:
Is there any reason to keep supporting python 3.5? The new dataset will be incompatible with python 3.5 (unless we do a fair bit of rewriting), and all new user are installing Anaconda with python 3.6 these days.

Does anyone know of a good reason to not move forward? Would it break something for anyone?

@QCoDeS/core (but I'd like to hear from everyone)

jenshnielsen, 2017-12-18:
👍  from me. I suggest raising this on slack too

peendebak, 2017-12-18:
@WilliamHPNielsen Many of our systems are still on python 3.5. I would like to keep qcodes compatible for some time if possible. Not sure if something would break, but I am not keen on trying.

WilliamHPNielsen, 2017-12-18:
@peendebak Good to know. I'm glad we're discussing this now, then. I think we are going to drop python 3.5 support as soon as we can, the question is only when that is. It would be natural to do it in connection with merging the dataset PR, which is currently believed to happen around the end of January. Is it unrealistic for you to transition over to python 3.6 in the course of one month?

WilliamHPNielsen, 2018-03-13:
This happened when #806 was merged. We no longer support python 3.5

closed 2018-03-13 by WilliamHPNielsen