microsoft/Qcodes #929: Import data from old QCoDeS dataset

jenshnielsen, 2017-12-19:


WilliamHPNielsen, 2017-12-19:
I think we should go by the following strategy:

- Write a function that loads snapshot + `.dat` files and generates a new run in the database with these data
- Make a simple API utilising this function for putting large amounts of (old style) datasets into a database

There is no general way to make the translation, since it's arbitrary what in the old scheme corresponds to experiments

jenshnielsen, 2018-01-17:
Did some experimentation in a notebook just put it here for reference

```
location = '../data/2018-01-17/#002_2D_test_15-43-14'
loaded_data = qc.load_data(location)

# add parameters
# this is missing the set points information at the moment
meas = Measurement()
for arrayname, array in loaded_data.arrays.items():
    meas.register_custom_parameter(name=array.name,
                                   paramtype='real',
                                   label=array.name,
                                   unit=array.unit
                                   )
# store all arrays that are not setpoints one by one
for arrayname, array in loaded_data.arrays.items():
    if not array.is_setpoint:
        store_array_to_database(arrayname, array)

def store_array_to_database(arrayname, array):
    print(f'got an array {arrayname}')
    dims = len(array.shape)
    if dims == 2:
        outer_data = np.empty(array.shape[1])
        for index, i in enumerate(array.set_arrays[0]):
            outer_data[:] = i
            print("x {} y {} z {}".format(outer_data, array.set_arrays[1][index,:], array[index,:]))
    elif dims == 1:
        pass
    elif dims == 0:
        pass
    else:
        raise NotImplementedError('dont know how to handle larger arrays')
```

jenshnielsen, 2018-01-17:
Need to figure out

* a general way of getting the dependencies (setpoints) that includes handling arrays of arbitrary sizes and datasets with multiple measured arrays with possibly different arrays
* Should it be possible to insert an array as multiple rows using the context manager api

WilliamHPNielsen, 2018-01-18:
> Should it be possible to insert an array as multiple rows using the context manager api

It currently happens by default (for sufficiently simple cases). It ain't documented, but see e.g. here: https://github.com/QCoDeS/Qcodes/blob/ae619137fc6819af175f3197aed858d605975645/qcodes/tests/dataset/test_measurement_context_manager.py#L349

jenshnielsen, 2018-01-18:
@WilliamHPNielsen Yes but that inserts it as a blob right?

WilliamHPNielsen, 2018-01-18:
No, it should unravel. Now (aa013f586dee0b9d5c53e578b3c9e703c53fb8d5) I changed the API to not worry about types and trying to unravel always.