microsoft/Qcodes #930: Finish Context manager

jenshnielsen, 2017-12-19:


WilliamHPNielsen, 2017-12-19:
We still need to settle on an API for adding QCoDeS parameters as well as custom parameters and to hit it heavily with tests.

WilliamHPNielsen, 2018-01-05:
Some work items:
- [ ] Switch methods names from camelCase to snake_case
- [ ] Make `add_result` call signature match the one one level deeper in the API
- [ ] Add `register_paramspec` function (to add custom parameter)
- [ ] Agree on measurement object hierarchy
- [ ] tests
- [ ] more tests

WilliamHPNielsen, 2018-01-22:
@QCoDeS/core I think the context manager is done, at least for a beta version. Comments and pointing out of blunders most welcome.

I'll put up an example notebook using some actual instruments including ones with `ArrayParameters`.

WilliamHPNielsen, 2018-07-23:
Closing this issue as it has outlived its purpose as a TO-DO point during a sprint.

closed 2018-07-23 by WilliamHPNielsen