microsoft/Qcodes #942: Dataset crashes when calling "add_results" if array is too large: "OperationalError: too many SQL variables"

sohailc, 2018-01-10:
The following code reproduces this error: 

```python
from qcodes import ParamSpec, new_data_set, new_experiment

new_experiment("bug", "buggy")
data_set = new_data_set("none")

x = ParamSpec('x', 'number', label='none', unit='none')
data_set.add_parameter(x)
result = [{"x": v} for v in range(1000)]  # This works with 999
data_set.add_results(result)
```

The issue appears to be in the "insert_many_values" function of the sqlite_base module. The smallest value where this issue appears is 1000. This agrees with the sqlite documentation here: http://www.sqlite.org/limits.html  (see Maximum Number Of Host Parameters In A Single SQL Statement)

A proposed solution is to add results to the database in chunks of sizes 999 or smaller.

WilliamHPNielsen, 2018-01-12:
Curiously, I can't reproduce it with your numbers. I ran

```python 
from qcodes import ParamSpec, new_data_set, new_experiment

new_experiment("bug", "buggy")
data_set = new_data_set("none")

x = ParamSpec('x', 'real', label='none', unit='none')
data_set.add_parameter(x)
Ns = [1000, 10000, 100000]  # I get an error with 1e6
for N in Ns:
    result = [{"x": v} for v in range(N)]
    data_set.add_results(result)
    print(f'The data set has {len(data_set)} points')
```
and I only get the error at one million. What's going on? My HEAD is at 47ef6bbb5e5bfa8e32f18f96b15f42e326ae11ca

sohailc, 2018-01-12:
Let me try things again after catching up with the latest commit.

sohailc, 2018-01-12:
the issue is still there for me, even after catching up with the latest commit of feature/dataset: 
https://github.com/QCoDeS/Qcodes/commit/b007c46f107e20cd9574e799d6513ff6df25e438

SQLITE_MAX_VARIABLE_NUMBER defaults to 999. Maybe its a different number of you? What is the error message you get with 1e6?

sohailc, 2018-01-12:
Your code fails strat away for me, at N=1000

WilliamHPNielsen, 2018-01-12:
b007c46 is 7 days old and by no means the latest commit in feature/dataset..?

I have no clue how to check the limits that sqlite on my machine has been compiled with. But it would be very helpful for us to know how to check for that. I'll keep googling it.

I got the same error message as you: "OperationalError: too many SQL variables".

I have added a test `test_add_parameter_values` to `test_dataset_basic.py` that tests the 999 limit, but it fails on travis (https://travis-ci.org/QCoDeS/Qcodes/builds/328100664?utm_source=github_status&utm_medium=notification) with N=501 and a different error message, "sqlite3.OperationalError: too many terms in compound SELECT". It passes locally on my machine.

-There is something fishy going on!

sohailc, 2018-01-12:
Oh maybe I got that number (b007c46) wrong, but I just pulled the feature/dataset branch from the main QCoDeS repo and merged it with my branch so I should have the latest code. 

I also updated my qcodes i environment by deleting the environment first (conda remove env -n qcodes) and recreating it (conda env create). Maybe I can try to explicitly update sqlite3 and see if that helps anything. 

I think this contains an interesting solution: https://stackoverflow.com/questions/35616602/peewee-operationalerror-too-many-sql-variables-on-upsert-of-only-150-rows-8-c

WilliamHPNielsen, 2018-01-12:
Nice link! It seems that we should indeed fetch all these limits from whatever machine we run on and use that information.

@jenshnielsen just helped me to view my compile options:
start sqlite3 with *no* database from the command line
```bash
sqlite3
```
and then
```sql
select sqlite3_compileoption_get(28);
```
echoes back: `MAX_VARIABLE_NUMBER=500000` on my machine.

sohailc, 2018-01-12:
Hmmm, I think you guys are on Linux/Mac right. I cannot find an executable sqlite3 on my system (I'm on windows 10)

jenshnielsen, 2018-01-12:
It's probably better to execute the statement from python. Something like the following should work

```python
import sqlite3
conn = sqlite3.connect('example.db')
c = conn.cursor()
c.execute('select sqlite_compileoption_get(29);')
c.fetchone()
('MAX_VARIABLE_NUMBER=500000',)
```
I am not sure that the ordering is stable so 29 may not be the correct number on your machine

WilliamHPNielsen, 2018-01-12:
Excellent. I volunteer to add the fetching all this info to figure out chunk sizes and other potentially critical variables.

sohailc, 2018-01-12:
Awesome! Thanks Jens. Let me report the output of that shortly

sohailc, 2018-01-12:
Yes I am getting None on my machine. Let me try to figure out the correct number for my machine.

sohailc, 2018-01-12:
I tried different number in a loop with the following result: 
[image]

I think the option MAX_VARIABLE_NUMBER as not set at compile time, so it is using the default value of 999 :-(

sohailc, 2018-01-12:
@WilliamHPNielsen Will you then create a PR for fixing this? Or should I work on the issue?

WilliamHPNielsen, 2018-01-12:
@sohail Neither :)

I'll fix this, but we just push everything related to the dataset into the existing PR #664. So no new PR (it's a bit messy, admittedly, but it's only for few weeks until the new dataset lands 😛 )

sohailc, 2018-01-12:
Ok cool, I will be looking forward to your fix :-)

WilliamHPNielsen, 2018-01-16:
Okay, I *think* it is properly fixed now (with "now" being 251a2fb48a3865b8a3f5b6898b13a93586cba3e7), but as you can see in the commit history, other ugly problems popped up. But now we have passing tests that explicitly go over the `MAX_VARIABLE_NUMBER` limit.

sohailc, 2018-01-16:
Is that the right link? I don't understand how increasing the deadline solves this problem

WilliamHPNielsen, 2018-01-16:
It is the right commit, but you'd have to look at the commit history to make sense of what is going on. 251a2fb is the first commit that passes the tests, but the real solution came earlier with the introduction of the `SQLiteSettings` object. 

We are riding dirty with one big PR for many different issues, therefore you need to dig a bit to see what's going on.

WilliamHPNielsen, 2018-01-31:
Imma mark this as closed, since we honestly think that we've solved the issue.

closed 2018-01-31 by WilliamHPNielsen