microsoft/Qcodes #967: driver/AMI430 broken by PR #954

Dominik-Vogel, 2018-02-07:
PR #954 breaks the ami @spauka @sohailc .
on qc 0.1.9:
```
ix = AMI430("x", address="192.168.15.121", port=7180)
ix.field(0)
```
leads to:
```
Traceback (most recent call last):

  File "<ipython-input-3-1db96f862287>", line 1, in <module>
    ix.field(0)

  File "c:\qcodes\qcodes\instrument\parameter.py", line 250, in __call__
    self.set(*args, **kwargs)

  File "c:\qcodes\qcodes\instrument\parameter.py", line 417, in set_wrapper
    raise e

  File "c:\qcodes\qcodes\instrument\parameter.py", line 399, in set_wrapper
    set_function(parsed_scaled_mapped_value, **kwargs)

  File "c:\qcodes\qcodes\utils\command.py", line 177, in __call__
    return self.exec_function(*args)

  File "c:\qcodes\qcodes\instrument_drivers\american_magnetics\AMI430.py", line 268, in set_field
    if np.abs(value) > self.field_rating():

  File "c:\qcodes\qcodes\instrument\parameter.py", line 244, in __call__
    return self.get()

  File "c:\qcodes\qcodes\instrument\parameter.py", line 351, in get_wrapper
    raise e

  File "c:\qcodes\qcodes\instrument\parameter.py", line 321, in get_wrapper
    value = get_function(*args, **kwargs)

  File "c:\qcodes\qcodes\utils\command.py", line 177, in __call__
    return self.exec_function(*args)

  File "c:\qcodes\qcodes\instrument_drivers\american_magnetics\AMI430.py", line 167, in <lambda>
    get_cmd=lambda: self.current_rating(),

  File "c:\qcodes\qcodes\instrument\parameter.py", line 244, in __call__
    return self.get()

  File "c:\qcodes\qcodes\instrument\parameter.py", line 351, in get_wrapper
    raise e

  File "c:\qcodes\qcodes\instrument\parameter.py", line 321, in get_wrapper
    value = get_function(*args, **kwargs)

  File "c:\qcodes\qcodes\utils\command.py", line 177, in __call__
    return self.exec_function(*args)

  File "c:\qcodes\qcodes\utils\command.py", line 126, in call_by_str
    return self.exec_str(self.cmd_str.format(*args))

  File "c:\qcodes\qcodes\instrument\base.py", line 648, in ask
    raise e

  File "c:\qcodes\qcodes\instrument\base.py", line 641, in ask
    answer = self.ask_raw(cmd)

  File "c:\qcodes\qcodes\instrument\ip.py", line 181, in ask_raw
    return self._recv()

  File "c:\qcodes\qcodes\instrument\ip.py", line 145, in _recv
    result = self._socket.recv(self._buffer_size)

timeout: ('timed out', "asking 'CURR:RATING?' to <AMI430: x>", 'getting x_current_rating', 'getting x_field_rating', 'setting x_field to 0')
```

sohailc, 2018-02-07:
I cannot seem to reproduce this with pyvisa-sim backend. Maybe the issue only occurs with an actual instrument?

sohailc, 2018-02-07:
I will run the QCodes example with AMI430 notebook in the documents with an actual magnet to see if everything still runs ok. I may not have access to a magnet today though. @spauka can you also look at this?

spauka, 2018-02-08:
Hey @Dominik-Vogel 

I can't seem to reproduce this on v0.1.9, or on master (e4986acdae801f1661dfc1e5519d9d16e3b88d93)

Could this be related to #966?

WilliamHPNielsen, 2018-04-12:
@spauka we are now seeing this on a fridge in Copenhagen. I'm shooting blindly here, but do your powers supply have the PSwitch option installed? Ours don't and I think it might be the cause.

WilliamHPNielsen, 2018-04-12:
Hmm... now I'm less sure about my blind guess. It seems more like a firmware version issue. There are several options from the manual missing in the onboard menus.

WilliamHPNielsen, 2018-04-12:
The power supply controlled by our AMI 430 programmer is a 4Q06125PS. @spauka, which one do you have?

spauka, 2018-04-13:
@WilliamHPNielsen I will double check the version string of the controller on Wednesday (currently out of the lab), but the power supply at least is the same (4Q06125PS).

I would be reasonably confident that it isn't the PSWITCH option as I checked that this disables correctly but the firmware version seems likely.

The other thing you could try is upgrade the controller firmware? From my reading of the manual the idea of the two commands was to be able to set a limit by the magnet construction (current rating) and a user limit (current limit) as an extra level of safety. But there's also no need for two things that end up doing roughly the same job, so #1052 would be a good fix if it works :)

PR #1052 (2018-04-17): Driver/ami430 current range
Fixes #967 by using `current_limit` instead of `current_rating` for checks. I think this might anyway be a better option.

@spauka and @sohailc, it would be nice if you could try this out on a real instrument (and please report your firmware revision!).

I'll try it out here in Copenhagen to see if my bold fixing claim is even true.

@Dominik-Vogel
closed 2018-04-17 by jenshnielsen