microsoft/Qcodes #1040: db_location config

robert-wallis, 2018-04-11:
Our exiting, working initialization code now encounters an error from a clean install without an environments.db file.

This may block Dave and Soheil in Delft tomorrow/tonight.

How can we initialize it correctly?

### Steps to reproduce
1. `from qcodes import config`
2. `from qcodes.dataset.data_export import get_shaped_data_by_runid`

### Expected behaviour
No error.

### Actual behaviour
`qcodes.config["core"]["db_location"]` is unset

`KeyError: 'db_location'`

[image]

### System
**operating system**
Windows 10

**qcodes branch**
1.10

**qcodes commit**
 497fa48 (release on pip)

jenshnielsen, 2018-04-11:
The config file bundled with Qcodes has that variable set https://github.com/QCoDeS/Qcodes/blob/master/qcodes/config/qcodesrc.json I suspect that you have another older file lying around somewhere. AFAIK it loads from user home directory as well as the one bundled in QCodes

jenshnielsen, 2018-04-11:
or in cwd

WilliamHPNielsen, 2018-04-11:
Yes, the hierarchy is

1. The `qcodesrc.json` in cwd
2. The `qcodesrc.json` in home
3. The `qcodesrc.json` in `qcodes/qcodes/config`

jenshnielsen, 2018-04-11:
It looks like the config file bundled in the whl binary package is out of date for some reason. I think we will have to do a new release. For now you can get the tar.gz by installing qcodes with `pip install --no-binary qcodes qcodes` which does not seems to have this issue

jenshnielsen, 2018-04-11:
or get the file json file (and schema) from the repo and put in user home or cwd

WilliamHPNielsen, 2018-04-11:
@robert-wallis I just put out a new release, v0.1.11. Does using that resolve your issue?

robert-wallis, 2018-04-11:
Yes @geoffroth just confirmed the 1.11 release fixed this. 🎉 

Thank you for the speedy release! 🙏

closed 2018-04-11 by robert-wallis

jenshnielsen, 2018-04-11:
Great for the record #1044 should make it easier to spot this kind of issue