microsoft/Qcodes #1048: Retire the Loop in the documentation

WilliamHPNielsen, 2018-04-12:
The first thing that a new user sees when stumbling across the QCoDeS documentation is how to set up a measurement with the old, hopefully soon-to-be-retired Loop. That should not be so. I propose that we shuffle around the documentation a bit and in particular make the `Tutorial.ipynb` and the `docs/index.rst` use the new dataset and context manager.

peendebak, 2018-04-12:
@WilliamHPNielsen Retirement of the `Loop` is great. I would make sure that in the updated docs it is mentioned that we have two datasets (`qcodes.data.data_set` and in `qcodes.dataset.data_set`) and both are valid to use.

WilliamHPNielsen, 2018-04-25:
In the longer term, I think we are not going to support the old dataset either. The current plan is to move the both the Loop and the old dataset into some legacy package that we do not actively maintain and may break compatibility with.