microsoft/Qcodes #1057: Dataset Context Manager notebook fails on Windows

WilliamHPNielsen, 2018-04-17:
The `Dataset Context Manager.ipynb` notebook randomly fails on Windows when executed from the command line. 

### Steps to reproduce
1. Run `jupyter nbconvert --to notebook --ExecutePreprocessor.timeout=600 --execute "Dataset Context Manager.ipynb"`
2. Repeat ~ 20 times

### Expected behaviour
The notebook should execute and the above command should exit with return code 0.

### Actual behaviour

#### Locally 
When running the following notebook on my local Windows machine:

```
from subprocess import run, PIPE

cmd = ['jupyter-nbconvert', '--to', 'notebook',
       '--execute', 'Dataset Context Manager.ipynb']

successes = []
fails = []
N = 100

for n in range(N):
    print(f'[*] Executing notebook, run {n}:{N-1}')
    
    result = run(cmd, stdout=PIPE, stderr=PIPE)
    if result.returncode == 0:
        print('[+] Succes')
        successes.append(result)
    else:
        print(f'[-] Fail, return code {result.returncode}')
        fails.append(result)
    print(f'Current score: {len(fails)} fails, {len(successes)} successes.')
    print('='*25)
```
I get the following error roughly 1 out of 11 times:
```
[NbConvertApp] Converting notebook Dataset Context Manager.ipynb to notebook
[NbConvertApp] Executing notebook with kernel: python3
[NbConvertApp] ERROR | Error while converting 'Dataset Context Manager.ipynb'
Traceback (most recent call last):
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\nbconvertapp.py", line 381, in export_single_notebook
    output, resources = self.exporter.from_filename(notebook_filename, resources=resources)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\exporters\exporter.py", line 172, in from_filename
    return self.from_file(f, resources=resources, **kw)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\exporters\exporter.py", line 190, in from_file
    return self.from_notebook_node(nbformat.read(file_stream, as_version=4), resources=resources, **kw)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\exporters\notebook.py", line 31, in from_notebook_node
    nb_copy, resources = super(NotebookExporter, self).from_notebook_node(nb, resources, **kw)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\exporters\exporter.py", line 132, in from_notebook_node
    nb_copy, resources = self._preprocess(nb_copy, resources)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\exporters\exporter.py", line 309, in _preprocess
    nbc, resc = preprocessor(nbc, resc)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\preprocessors\base.py", line 47, in __call__
    return self.preprocess(nb,resources)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\preprocessors\execute.py", line 242, in preprocess
    nb, resources = super(ExecutePreprocessor, self).preprocess(nb, resources)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\preprocessors\base.py", line 70, in preprocess
    nb.cells[index], resources = self.preprocess_cell(cell, resources, index)
  File "C:\Users\qcodes-william\Anaconda3\envs\qcodesqcodes\lib\site-packages\nbconvert\preprocessors\execute.py", line 275, in preprocess_cell
    raise CellExecutionError(msg)
nbconvert.preprocessors.execute.CellExecutionError: An error occurred while executing the following cell:
------------------
ax, cbax = plot_by_id(run_id)
------------------

IndexError: index 0 is out of bounds for axis 0 with size 0
```

Note: the same is seen with a simple batch file running a loop that executes the notebook over and over. Running the same python code on a mac seems to always succeed.

#### AppVeyor

On AppVeyor, we sometimes (I don't have a good number, perhaps 1 in 5 times?) get **another** error from the same notebook execution:
```
---------------------------------------------------------------------------
OperationalError                          Traceback (most recent call last)
<ipython-input-30-42e08b26d127> in <module>()
     20 meas.add_subscriber(get_list_of_first_param, state=first_param_list)
     21 
---> 22 with meas.run() as datasaver:
     23     for n in range(10):
     24         datasaver.add_result((dac.ch1, n), (dmm.v1, n**2))
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\measurements.py in __enter__(self)
    254             # to the database
    255             log.debug(f'Subscribing callable {callble} with state {state}')
--> 256             self.ds.subscribe(callble, min_wait=0, min_count=1, state=state)
    257 
    258         print(f'Starting experimental run with id: {self.ds.run_id}')
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\data_set.py in subscribe(self, callback, min_wait, min_count, state, callback_kwargs, subscriber_class)
    550         sub_id = hash_from_parts(str(time.time()))
    551         sub = Subscriber(self, sub_id, callback, state, min_wait, min_count,
--> 552                          callback_kwargs)
    553         self.subscribers[sub_id] = sub
    554         sub.start()
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\data_set.py in __init__(self, dataSet, sub_id, callback, state, min_wait, min_count, callback_kwargs)
    101             SELECT {self.callbackid}({param_sql});
    102         END;"""
--> 103         atomicTransaction(self.conn, sql)
    104         self.data: Queue = Queue()
    105         self._data_set_len = len(dataSet)
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\sqlite_base.py in atomicTransaction(conn, sql, *args)
    258         logging.exception("Could not execute transaction, rolling back")
    259         conn.rollback()
--> 260         raise e
    261 
    262     conn.commit()
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\sqlite_base.py in atomicTransaction(conn, sql, *args)
    254     """
    255     try:
--> 256         c = transaction(conn, sql, *args)
    257     except Exception as e:
    258         logging.exception("Could not execute transaction, rolling back")
C:\Miniconda36-x64\envs\qcodes\lib\site-packages\qcodes\dataset\sqlite_base.py in transaction(conn, sql, *args)
    233         c.execute(sql, args)
    234     else:
--> 235         c.execute(sql)
    236     return c
    237 
OperationalError: trigger sub1837cafd1570f508b3c10e6a23eceed41d552679 already exists
OperationalError: trigger sub1837cafd1570f508b3c10e6a23eceed41d552679 already exists

```
### System
**Windows**

**master**

**0c3cd4c**

@QCoDeS/core

jenshnielsen, 2018-04-17:
I think the problem is that the subscriber is generated with an id which is `hash_from_parts(str(time.time()))` I suspect that in tests this this may be called to fast after each other and we get two subscribers with the same id. I suggest adding some random numbers to the time or use the uuid module

jenshnielsen, 2018-04-17:
The output of the following on appveyor could be interesting:

```python
In [10]: for clock in ['clock', 'monotonic', 'perf_counter', 'process_time', 'time']:
    ...:     print(f'{clock}: {time.get_clock_info(clock)}')
    ...:
clock: namespace(adjustable=False, implementation='clock()', monotonic=True, resolution=1e-06)
monotonic: namespace(adjustable=False, implementation='mach_absolute_time()', monotonic=True, resolution=1e-09)
perf_counter: namespace(adjustable=False, implementation='mach_absolute_time()', monotonic=True, resolution=1e-09)
process_time: namespace(adjustable=False, implementation='clock_gettime(CLOCK_PROCESS_CPUTIME_ID)', monotonic=True, resolution=1.0000000000000002e-06)
time: namespace(adjustable=True, implementation='clock_gettime(CLOCK_REALTIME)', monotonic=False, resolution=1.0000000000000002e-06)
```

WilliamHPNielsen, 2018-04-17:
@jenshnielsen, yes, we should probably look into timings on appveyor. Here is another timing problem https://ci.appveyor.com/project/WilliamHPNielsen/qcodes/build/1.0.68

jenshnielsen, 2018-04-18:
Pushed #1063 to get some insight. Once we can require python 3.7 we should use `time.monotonic_ns()` https://docs.python.org/3.7/library/time.html#time.monotonic_ns but that is a way off

jenshnielsen, 2018-04-18:
This https://msdn.microsoft.com/en-us/library/windows/desktop/dn553408(v=vs.85).aspx also states that the perf counter is monotonic on windows

jenshnielsen, 2018-04-18:
Python monotonic cloc uses [this](https://msdn.microsoft.com/da-dk/library/windows/desktop/ms724411(v=vs.85).aspx) for some reason

WilliamHPNielsen, 2018-04-20:
Reopening this since #1065 is needed as well to fully alleviate this.

closed 2018-04-23 by jenshnielsen