microsoft/Qcodes #1071: Dataset / agree on what `load_by_name`

WilliamHPNielsen, 2018-04-25:
In the new dataset, several people have already been bugged by the facts that

- An experiment and a run can have any non-unique name
- `load_by_name` expects a unique name

(see e.g. #991)

I think we have an API in conflict with itself, which I suggest that we change. There are two obvious ways to go (with the first way branching into two ways). Either 

 - An experiment and a run is forced to have a unique name (in one of two ways)
   - This happens by raising an error when a name i reused
   - This happens by automatically changing the name to a unique one (prepending a number or so)
 - `load_by_name` expects a unique name

or

 - An experiment and a run can have any non-unique name
 - `load_by_name` loads all experiments/runs that match the given name

So, which one do you prefer? 1a, 1b, or 2?

@QCoDeS/core @jana-d @majacassidy

Dominik-Vogel, 2018-04-25:
By looking at how the old dataset gets used here at some stations in Copenhagen, I would like to add a third level of identification, i.e. in addition to `name` and `sample` , adding a string field like `series`.

WilliamHPNielsen, 2018-04-25:
@Dominik-Vogel, perhaps you can elaborate a bit on that? Do you mean for the experiments? How is it they use the old dataset and what are we curerntly missing?

WilliamHPNielsen, 2018-04-25:
I see. But how is that not captured in the name of the experiment?

jana-d, 2018-04-25:
I agree with William that the experiment name should/could be enough to specify everything.

Dominik-Vogel, 2018-04-25:
Thanks @jana-d . I went around to talk to some experimenters about that an found that most would be happy with the level of nesting we have. Additionally in these conversations the following came up:
- Samples do get reloaded and it might be useful to mark that in the data
- It might be desirable to have a association between multiple users and runs.

jana-d, 2018-04-25:
As for the options for the name problem, I vote for 1a). 
Throw an error when trying to create an experiment that already exists. Simply to make the user aware that he/she is doing this, which might be unconscious. The user can then still pre or append whatever is needed to make the name unique.

spauka, 2018-05-07:
Agreed with jana, option 1a).

Automatically appending things to names ends up being messy, particularly if the intention is for the experiment to be the same!