microsoft/Qcodes #1108: register_parameter always assumes 'numeric' type

sohailc, 2018-05-25:
Register parameter always uses paramtype='numeric' when creating the ParamSpec, regardless of the type of parameter to be registered. This is problematic when we try to register array-valued parameters.

WilliamHPNielsen, 2018-05-30:
I think this is indeed a bug, but let me add a bit of background before we turn everything back into arrays.

At some point we agreed to convert ("unravel" was the word, if I remember correctly) arrays into numbers to avoid having binary blobs in the database. For a 2D sweep of, say, `V1` and `V2` with a signal `Y`, that entails converting what would be a table

 V1 | V2 | Y
----|-----|-------
   0  |   BLOB  | BLOB
   1  |  BLOB  | BLOB

into a table more like

 V1 | V2 | Y
----|-----|-------
   0  |  0  | 0.123
   0  |  1  | 0.234
   0 |   2  |  1.342
   1 |  0  |  -.231

That's probably good to keep in mind when solving this issue.

nataliejpg, 2018-08-06:
I would also add to this that MultiChannel parameters dont currently work with add_parameter, also perhaps this is written somewhere but is there a function to get the data out of the database if it is saved as a binary blob? get_data and get_values didn't work for me? mention of both also in #1200

PR #1207 (2018-08-17): Dataset Context manager. Allow insertion of Data direcly as arrays
This is significantly faster than unrolling and inserting point by point. This in turn uncovered some other issues.

* Correct insertion of multidimensional Arrays in the context manager. Previously it would only 
unroll the first axis resulting in inconsistent results. 
* Adapt the data exporter `get_data_by_id` to return full arrays of setpoints regardless of the way they are inserted.

TODO:

- [x] Add Benchmarks of the difference vs insertion as single points
- [ ] The tests have a fair bit of unneeded code duplication
- [x [... cut]
closed 2018-08-17 by jenshnielsen