microsoft/Qcodes #1118: Cannot set QCoDeS Instrument Parameter with numpy.int32, int16, int8

mwilmer, 2018-06-03:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.

qcodes issue:
Cannot set QCoDeS Instrument Parameter with numpy.int32, int16, int8 etc

### Steps to reproduce
1. create/connect to a real instrument (here: a QuTech D5a dac module)
2. try to set a value using a numpy int32 -> my_instrument.my_parameter(np.int31(1))
3. 

### Expected behaviour
Instrument should be set normally (works with setting with normal floats, ints, or numpy floats)

### Actual behaviour
Nothing is set, following warning is printed. No exception is thrown. Response seems to be asynchronous.

`WARNING:root:cannot sweep dac9 from 9.002822919050914 to 1 - jumping`

Setting several values repeatedly in a loop causes the warnings to be printed in groups/ bursts.

### System
**operating system**
Windows 10

**qcodes branch**
v0.1.11 via pypi

**qcodes commit**

WilliamHPNielsen, 2018-06-04:
Thanks for reporting this issue. The example you provide actually does not reproduce the warning, as it is missing one ingredient: setting the step value.

The following produces no warning:
```python
import numpy as np
import qcodes as qc
from qcodes.tests.instrument_mocks import DummyInstrument

instrument = DummyInstrument('instrument', gates=['gate1'])
for typ in [np.int64, np.int32, np.int16, np.int8]:
    instrument.gate1(9)
    instrument.gate1(typ(1))
```
whereas this does:
```python
import numpy as np
import qcodes as qc
from qcodes.tests.instrument_mocks import DummyInstrument

instrument = DummyInstrument('instrument', gates=['gate1'])
instrument.gate1.step = 1
instrument.gate1(1)
instrument.gate1(np.int32(1))
```

The problem is in `get_ramp_values` that should probably just be made more permissive.

WilliamHPNielsen, 2018-06-04:
@nulinspiratie, do you see any reason to only allow `int` and `float`?