microsoft/Qcodes #1121: loading dataset with new dataset code fails

peendebak, 2018-06-04:
Importing a dataset with `import_dat_file` fails for simple datasets.

A minimal example:
```
import os
from qcodes.dataset.legacy_import import import_dat_file

from qcodes.dataset.database import initialise_database
from qcodes.dataset.experiment_container import new_experiment

import numpy as np
import qcodes.tests
import qcodes.tests.data_mocks

# create dataset
ds=qcodes.tests.data_mocks.DataSet2D()
ds.write()
location=ds.location

initialise_database()

# put the old data in a new experiment
new_experiment('old_data_loading', sample_name='no_sample')
location1d = os.path.join(qcodes.DataSet.default_io.base_location, location)
run_ids = import_dat_file(location1d)
```
The code fails with error `AttributeError: 'NoneType' object has no attribute 'isidentifier'`
@WilliamHPNielsen @jenshnielsen @sohailc

jenshnielsen, 2018-06-05:
The example notebook works as expected.

Without a few more details it's hard to figure out whats going on. Could you provide a full traceback so we can see where that error comes from. Can you check that the location1d actually points to a folder congaing dat files

peendebak, 2018-06-05:
@jenshnielsen The full traceback is:
```
n [5]: Traceback (most recent call last):

  File "<ipython-input-4-0bbd4f2af0c8>", line 21, in <module>
    run_ids = import_dat_file(location1d)

  File "c:\svn\qcodes\qcodes\dataset\legacy_import.py", line 75, in import_dat_file
    meas = setup_measurement(loaded_data)

  File "c:\svn\qcodes\qcodes\dataset\legacy_import.py", line 26, in setup_measurement
    setpoints = setarrays

  File "c:\svn\qcodes\qcodes\dataset\measurements.py", line 505, in register_custom_parameter
    depends_on=depends_on)

  File "c:\svn\qcodes\qcodes\dataset\param_spec.py", line 30, in __init__
    if not name.isidentifier():

AttributeError: 'NoneType' object has no attribute 'isidentifier'
```
The `location1d` is indeed a folder on disk. Does the minimal example work for you?

jenshnielsen, 2018-06-05:
Yes for some reason that .dat file has much less metadata when loaded than the dataset the example uses. I am not yet sure why. The files look the same to me

jenshnielsen, 2018-06-05:
This looks like to be due to a missing snapshot that contains metadata about the array relations

peendebak, 2018-06-05:
@jenshnielsen The `import_dat_file` does work when I add `qcodes.DataSet.default_formatter=qcodes.data.hdf5_format.HDF5Format()` on top of the script. So it is relate to the `GNUPlotFormatter`.

jana-d, 2018-06-15:
@jenshnielsen, indeed, the problem was the missing json snapshot. Came across the same error, pointing to a folder which had both the .dat and snapshot worked without adding the HDF5Format line above.