microsoft/Qcodes #1123: environment.yml and requirements.txt are not consistent

astafan8, 2018-06-05:
The Python environment settings are not consistent between environment.yml and requirements.txt. Note that the environment.yml is used by AppVeyor CI and requirements.txt is used by Travis CI (is it also an inconsistency that needs to be solved?)

So far as I understand, conda and pip(+virtualenv) are both quite popular, but qcodes developers prefer using conda, while users tend to use just pip.

I propose to:
- [ ] update requirements.txt to match environment.yml
- [ ] add a note somewhere (where?) that BOTH requirements.txt and environment.yml need to be updated when qcodes dependencies change (perhaps, implement this simple check it and add its execution to the CI systems)

jenshnielsen, 2018-06-05:
Yes we need to improve the dependency handling this has been bugging me for a while. There are issues somewhat outside our control that makes this harder. See some of my random notes below.

Note that #1120 syncs all the dependencies to the latests versions of various packages. 

Many of these annoyances are simply due to differences between how Anaconda and pip works. 
For example Anaconda up till very recently only packaged pyqt 5.6 but that version does not actually have working whl packages for python 3.6 on pypi. This is fixed in #1120 now that conda packages pyqt5.9.x. In the meantime pyqt5.10 has come out on pypi so the problem may arise again. 

> So far as I understand, conda and pip(+virtualenv) are both quite popular, but qcodes developers prefer using conda, while users tend to use just pip.

This is not that simple. It is more that conda is easier to get started with and is much easier to use on windows. On other platforms this matters much less. All users that we are aware of in Copenhagen and Sydney use conda. 

> update versions in requirements.txt to match versions in environment.yml
Neither actually does only list strict requirements. The strict requirements are defined in setup.py and installed automatically when you install qcodes. There are a bunch of optional dependencies for plotting

> so far as I see, environment.yml also contains info on packages that required only for testing; so let's also make environment.yml consistent with test_requirements.txt
This would effectively mean that those packages are not installed by conda but by pip when using 

It would be preferable to have a pure Conda environment but that is not possible (easily) because we 
depend on packages that are not packages for conda yet and may never be. I would like to find a way of having conda install from the x_requirements.txt files falling back to pip for packages that do not exist but I don't see a good way of doing this without our own scripts. 

We have considered packaging qcodes up for conda which would solve many of these problems but that is not really useful as far as I can see as long as many of our users tend to use non released versions from git checkouts. There is AFAIK not a simple way of conda installing a folder

astafan8, 2018-10-11:
Many improvements have been made already to this. One of the important decisions were to install everything via `conda` by default, hence there is no more requirement that "the users should be __equally__ able to set up qcodes environment with conda and pip". To conclude, this issue can be closed now.

closed 2018-10-11 by astafan8