microsoft/Qcodes #1125: Nonetype not accepted as valid type by data saver

sohailc, 2018-06-07:
Let's consider the situation where we interleave a 1D and 2D sweep. This will look as follows.  

```python
import numpy as np

from qcodes.dataset.measurements import Measurement
from qcodes import Parameter

x = Parameter("x", set_cmd=None, get_cmd=None)
y = Parameter("y", set_cmd=None, get_cmd=None)

m = Parameter("m")
m.get = lambda:np.sin(x())

n = Parameter("n")
n.get = lambda:np.sin(x()**2 + y())

meas = Measurement()
meas.register_parameter(x)
meas.register_parameter(y)
meas.register_parameter(m, setpoints=[x])
meas.register_parameter(n, setpoints=[x, y])

with meas.run() as datasaver: 
    for ix in np.linspace(0, 6, 100): 
        x.set(ix)
        datasaver.add_result(("x", ix), ("y", None), ("m", m.get()), ("n", None))

        for iy in np.linspace(0, 6, 100): 
            y.set(iy)
            datasaver.add_result(("x", ix), ("y", iy), ("m", None), ("n", n.get()))
```

Currently, this fails because of `ValueError: Wrong value type received. Got <class 'NoneType'>, but only int, float, str, tuple, list, and np.ndarray is allowed.`.

WilliamHPNielsen, 2018-06-07:
I'd actually be tempted to call this a feature rather than a bug. At least in the above example... What does the user gain from specifying the `None` values?

sohailc, 2018-06-07:
If the None value is not there then there is not the same number of columns at each insertion. This will also give an error. This is definitely a bug @WilliamHPNielsen

WilliamHPNielsen, 2018-06-07:
> If the None values are not there then there are not the same number of columns at each insertion.

That is a very good point and certainly something to fix, but that does not vote neither for nor against having users explicitly specifying `None` values.

I guess my only concern is that something like
```
meas = Measurement()
meas.register_parameter(x)
meas.register_parameter(y)
meas.register_parameter(z)

with meas.run() as datasaver: 
   datasaver.add_result((x, 0))
   datasaver.add_result((x, 0), (y, None))
   datasaver.add_result((x, 0), (z, None))
   datasaver.add_result((x, 0), (y, None), (z, None))
```
where the four last lines do the same thing is perhaps (?) a potential source of confusion.

I suppose there are three possible ways to go:

1. Maintaining that `None` values can only be added implicitly by their omission.
2. Allowing for both omitting a value and explicitly declaring that it is `None`.
3. Only allowing explicitly passing `None` values. 

Both 1. and 2. would require some extra work to fix the column mismatch bug that Sohail pointed out. I'd vote for either 1. or 2.

sohailc, 2018-06-07:
Let's go for option 1. I will update the PR. @jenshnielsen @Dominik-Vogel do you have any input?

sohailc, 2018-06-07:
@WilliamHPNielsen Can you have another look at the PR? You should now be able to do the following: 

```python
x = Parameter("x", set_cmd=None, get_cmd=None)
y = Parameter("y", set_cmd=None, get_cmd=None)

m = Parameter("m")
m.get = lambda:np.sin(x())

n = Parameter("n")
n.get = lambda:np.sin(x()**2 + y())

meas = Measurement()
meas.register_parameter(x)
meas.register_parameter(y)
meas.register_parameter(m, setpoints=[x])
meas.register_parameter(n, setpoints=[x, y])

with meas.run() as datasaver: 
    for ix in np.linspace(0, 6, 100): 
        x.set(ix)
        datasaver.add_result(("x", ix), ("m", m.get()))

        for iy in np.linspace(0, 6, 100): 
            y.set(iy)
            datasaver.add_result(("x", ix), ("y", iy), ("n", n.get()))
            
plot_by_id(datasaver.run_id)
```

sohailc, 2018-06-07:
No hang on, I just realized my implementation is *not* right. It assumes all keys are present when data is flushed to the database. If this is not true, this method may fail. Let me investigate

sohailc, 2018-06-07:
After testing a bit, it seems that my current implementation is working :-)

PR #1126 (2018-06-11): Nonetype accepted by datasaver
Fixes #1125 .
closed 2018-06-11 by astafan8