microsoft/Qcodes #1140: Inconsistent naming of parameters and instruments

jenshnielsen, 2018-06-12:
This is not so much a bug as an annoying inconsistency. I am not sure if there is a good way to fix this whire retaining back wards compatibility but none the less the naming is inconsistent consider the following:

| type       | name                                       | full_name                | short_name        |
|------------|--------------------------------------------|--------------------------|-------------------|
| Instrument | "{name}" e.g. "myinst"                       | N/A                      | N/A               |
| Channel    |  "...{parent_name}_{name}" e.g. "myinst_ch1" | N/A                      | "{name}" e.g. "ch1" |
| Parameter  | "{name}" e.g. "frequency"                  | "{instrument_name}_{name}" e.g. "{myinst_ch1_frequency}" | N/A               |

jenshnielsen, 2018-06-12:
We could make this better by always implementing short name and full_name:

* For Instrument they would all be the same 
* For Channels name and full_name would be the same
* For Parameter name and short_name would be the same

jenshnielsen, 2018-06-12:
@QCoDeS/core

WilliamHPNielsen, 2018-06-12:
Thanks for bringing this up. I have been annoyed by this confusion too.

I like you suggestion. What should `__str__` return?

jenshnielsen, 2018-06-12:
Perhaps
```
[self.name, self.full_name, self.short_name][random.randint(0,2)]
```
😈

jenshnielsen, 2018-06-12:
All stupid jokes aside. I will add that to the list above

jenshnielsen, 2018-06-12:
IMHO we should just use `__str__` as a human readable repr and not rely on any particular format of that

Dominik-Vogel, 2018-06-13:
Thank you @jenshnielsen , this has also been bugging me for some time. I like your suggestions. In an ideal world I would want to only have `full_name` and `name`  where `name` is what you suggest to be `short_name`. But I think your suggestion is a good way to keep compatibility.
We could consider deprecating `name` to establish a less confusing situation in the future.
I agree that `__str__` should be the full human readable version and not be used for logic. So ideally we  implement the parameter __str__ in the style of the channel version (e.g. parallel to `full_name`)

jenshnielsen, 2018-06-13:
@Dominik-Vogel sounds good to me

I also considered implementing something like a `name_parts` function that returns all the elements of the name in a list. For the parameter grouping it is useful to be able to get a name like channel_parameter but without the full instrument name included

Dominik-Vogel, 2018-06-13:
@jenshnielsen I like the `name_parts` idea. Maybe we can even pull all that code into a common place as a mixin?

WilliamHPNielsen, 2018-06-13:
> I agree that __str__ should be the full human readable version and not be used for logic.

Then I guess `measurements.py` needs some small fixes, since `str` is being used a lot there, also for logic.

jenshnielsen, 2018-06-13:
ok in that case it should be fine but also solvable by a straightforward replacement

jenshnielsen, 2018-06-13:
Reopening as we should still fix multi and array parameters to conform to the same logic