microsoft/Qcodes #1141: [desc to be added ] SweepMeasurement better design

astafan8, 2018-06-13:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### System
**operating system**

**qcodes branch**

**qcodes commit**

\

+class SweepMeasurement(Measurement):
+    def register_sweep(self, sweep_object: BaseSweepObject) -> None:
+
+        sweep_object.parameter_table.resolve_dependencies()
+        param_specs = sweep_object.parameter_table.param_specs
+
+        self.parameters = {
+            p.name: p for p in param_specs
+        }

WilliamHPNielsen, 2018-06-13:
@astafan8???

astafan8, 2018-06-13:
@WilliamHPNielsen I was writing the description, but something went wrong and the text is gone. I'm writing it again, now in smaller steps ;)

WilliamHPNielsen, 2018-06-14:
I fully agree. The `run` part has come up in discussions before, and it does seem a bit pointless to not just let the sweep do its thing with a single method.