microsoft/Qcodes #1151: Extract doc strings from dynanically added paramters

jenshnielsen, 2018-06-17:
Currently all parameters are dynamically added to instruments. This means that the autodoc documentation of the paramters does not contain the parameters. It would be great if we can find a way to include these docstrings in the autodoc generated documentation

caenrigen, 2021-05-27:
Hi, @jenshnielsen 

Is there any solution to this problem so far? And perhaps to as well https://github.com/QCoDeS/Qcodes/issues/32 that is somewhat related when, e.g.,  documenting subclasses of `Instrument` s

jenshnielsen, 2021-05-27:
Not at the moment. There does not seem to be a general solution to this in Sphinx. I think the best option would be to replace the use of add_parameter with an actual class instantiated as a member of the class. We are not actively working on this

With regards to #32 there was a suggestion in https://github.com/QCoDeS/Qcodes/pull/2816 unfortunately that hit some design issues

caenrigen, 2021-05-28:
@jenshnielsen thank you for the reply and information!

After several hours of prototyping I came up with a solution that solves #32 (and  probably #2816 as well).

At high level it involves:

- Extra (non-breaking) functionality in the base `Instrument` (or `_BaseInstrumet`) class.
- A relatively simple but specific way of adding parameters to instrument classes (older drivers could be upgraded to benefit from this as well).
- A bit of `sphinx` black magic.

## General idea

- Store the information needed to instantiate the parameters at class creation time --> use a dummy private method
- The dummy private method name and docstring will serve the creation of the `Parameter` s
- Use decorator to flag method for "conversion" to parameter.
- Sphinx is able to pick up the docstring as well --> autodoc possible.
- "Hack" sphinx to replace method name with the name of the parameter added at Intrument's instantiation time.
- Glue everything together.

## Sphinx docs result

[image]

## Usage: Instruments instantiation

```python
from quantify.utilities.my_mod import MyInstrumentDriver, SubMyInstrumentDriver

instr = MyInstrumentDriver("instr")
instr.freq(10)
instr.print_readable_snapshot()
print("\ninstr.time.label: ", instr.time.label)
```

```bash
instr:
	parameter value
--------------------------------------------------------------------------------
IDN  :	None 
freq :	10 (Hz)
time :	3 (s)

instr.time.label:  Time
```

```python
sub_instr = SubMyInstrumentDriver("sub_instr")
sub_instr.time(sub_instr.time() * 2)
sub_instr.print_readable_snapshot()
print("\nsub_instr.time.label: ", sub_instr.time.label)
```

```bash
sub_instr:
	parameter value
--------------------------------------------------------------------------------
IDN  :	None 
freq :	None (Hz)
time :	14 (s)

sub_instr.time.label:  Time long
```

## Usage: writing drivers

```python
from qcodes import Instrument, ManualParameter, Parameter
from qcodes import validators

class MyInstrumentDriver(ModifiedInstrumentClass):
[... 157 lines cut ...]
- everything works automatically
- documenting works in sphinx, including the extra arguments used to create the parameters, besides the docstring
- backwards compatible, i.e. does not break QCoDeS
- inheritance with overridden `Parameter` s seems to work from a few simple tests
- with some extra efforts more sphinx magic can probably be applied to create custom output if we would like to differentiate it more from what looks like a function

Cons: 

- requires extending `qcodes.Instrument` class or `qcodes._BaseInstrument`
- requires using a new custom way of adding parameters (I would say the `self.add_parameter` is already a custom one anyway)
- ... other issue???

## Questions

- What do you think about this?
- Is this approach clashing with anything in the QCoDeS repo?
- Would a MR request with these changes be accepted?

If yes:

- Should this become the recommended way of writing drivers?
- Where should the `_add_params_from_decorated_methods` be places (and executed)?
- where should the `add_parameter` decorator utility be placed?

Looking forward to your feedback, we ([at Quantify](https://gitlab.com/quantify-os/quantify-core)) would be happy to contribute with this feature.

caenrigen, 2021-06-03:
Hi!

@jenshnielsen, any advice on this?
In case you do not have time, is there some maintainer that could give some feedback?

Cheers

astafan8, 2021-06-03:
@caenrigen pardon for the delay.

Thank you so much for such a comprehensive and clear description of the suggestion!

There is a sense of hackiness in the solution, both in qcodes and in sphinx, however the implementation is short and is readable (which is very important). Another important aspect is that the solution does not break the existing way of specifying a parameter, and can live next to it -- that is important because I can't see ``InstrumentBase.add_parameter`` going away anytime soon.

I think we should try to see how it works in a PR to qcodes, and once integrated let's how it works for the quantify project :)

It's important to include the documentation of this "feature". it could be a separate example notebook, or could be just a section in the "writing drivers" notebook. I also think that splitting the feature into two would be beneficial for readers - one is "a new way of defining parameters" and two is "..that gives the benefit of addings parameters to autodoc in sphinx" - would you agree?

there were also specific questions:

> Should this become the recommended way of writing drivers?

unlikely. there are many cases where arguments to ``InstrumentBase.add_parameter`` are not "literally" known and need to be pre-calculated (perhaps even based on arguments to ``__init__``), and the "dummy private method with the add_parameter decorator" will not cover those.

now that i wrote this, perhaps, an adjustment to the solution which makes things better is that ``InstrumentBase.add_parameter`` defines the dummy private method with correct arguments, docstring, and NotImplementedError?

> Where should the _add_params_from_decorated_methods be places (and executed)?

in the init of ``InstrumentBase``.

> where should the add_parameter decorator utility be placed?

in ``qcodes.instrument.base`` module should be fine.

caenrigen, 2021-06-03:
@astafan8 thank you for the reply, we will follow up with a PR with this feature, might take a bit of time, but we certainly want this functionality.

> It's important to include the documentation of this "feature". it could be a separate example notebook, or could be just a section in the "writing drivers" notebook. I also think that splitting the [...]

agreed, there are at least two distinct functionalities in there

> there are many cases where arguments to InstrumentBase.add_parameter are not "literally" known and need to be pre-calculated (perhaps even based on arguments to __init__), and the "dummy private method with the add_parameter decorator" will not cover those.

I realized this myself as well after the proposal. Considering that the parameters are added only during `__init__()` I think it is possible to improve this a bit so that any argument to `add_paramter` (except the parameter name) can be a member of the (partially) initialized object. In the code above it is already being done to specify, e.g., the `set_cmd`.

> now that i wrote this, perhaps, an adjustment to the solution which makes things better is that InstrumentBase.add_parameter defines the dummy private method with correct arguments, docstring, and NotImplementedError?

I did not follow this part.

Perhaps related note: `InstrumentBase.add_parameter` currently is called from an instance object while the `@add_parameter` is executed when the module of `MyInstrumentDriver` is imported. In order to be able to document parameters in sphinx the information needs to be available when the `MyInstrumentDriver` is imported (sphinx imports all modules and then inspect the imported objects). Maybe this helps to understand some of the nuances.

>> Where should the `_add_params_from_decorated_methods` be places (and executed)?
> in the init of `InstrumentBase`.

now that we mention the parameters that require information that is available only after calling `__init__` I think it might be better to give users (driver writers) the responsibility of calling `add_params_from_decorated_methods`, so that some necessary information can be passed to `__init__`. (Some code snippets should make this more clear later).

astafan8, 2021-06-03:
> we will follow up with a PR with this feature, might take a bit of time, but we certainly want this functionality

no worries! looking forward to it!

> Perhaps related note: InstrumentBase.add_parameter currently is called from an instance object while the @add_parameter is executed when the module of MyInstrumentDriver is imported. In order to be [...]

ah, indeed! sorry, then please ignore that "now that i wrote this, perhaps, an adjustment to the solution" part.

> I think it might be better to give users (driver writers) the responsibility of calling add_params_from_decorated_methods, so that some necessary information can be passed to __init__. (Some code snippets should make this more clear later).

For the first PR I'd suggest to start with a private `_add_params_from_decorated_methods`, as it can always be made public in the future (while vice versa requires deprecation procedure). But let's also have some examples of how this responsibility would work.

jenshnielsen, 2021-10-14:
Fixed by https://github.com/QCoDeS/Qcodes/pull/3224 parameters should be rewritten using self.parameter_name = Parameter(...) for this to take effect

closed 2021-10-14 by jenshnielsen