microsoft/Qcodes #1152: Validation of non-required elements by qcodesrc_schema.json causes silent failiure of schema loading

Dominik-Vogel, 2018-06-18:
The current schema vaildation file `qcodes_schma.json` includes the validation of the station configurator object, but does not include it into the required list:
```
...
 "user": {
            "type": "object",
            "properties": {
                "scriptfolder": {
                    "type": "string",
                    "default": "/",
                    "description": "Location of scripts for general experiments"
                },
                "mainfolder": {
                    "type": "string",
                    "default": "",
                    "description": "Location of experiments"
                }
            },
            "description": "controls user settings of qcodes"
        },
        "station_configurator": {
            "type": "object",
            "properties": {
                "enable_forced_reconnect": {
                    "type": "boolean",
                    "default": false,
                    "description": "if set to true, on instantiation of an existing instrument, the existing will be disconnected."
                },
                "default_folder": {
                    "type": ["string", "null"],
                    "default": null,
                    "description": "default folder where to look for a station configurator config file"
                },
                "default_file": {
                    "type": ["string", "null"],
                    "default": null,
                    "description": "default file name, specifying the file to load, when none is specified. Can be a relative or absolute path"
                }

            },
            "description": "Setting for the StationConfigurator module in qdev-wrappers"
        }
    },
    "description": "schema for a qcodes config file",
    "required": [
        "gui",
        "core",
        "user"
    ]
}
```
This somehow prevents the config for the StationConfigurator to be loaded.
```
 from qdev_wrappers.station_configurator import StationConfigurator
sc = StationConfigurator()
```
here the default file as configured is not loaded.
When adding the SC to the required list the default file is being loaded successfully.

jenshnielsen, 2018-09-06:
@Dominik-Vogel is this still relevant?

Dominik-Vogel, 2018-09-06:
Yes and no. Yes in the sense that the behaviour has not changed and no in the sense that we are going to merge the station configurator into core.
This has low urgency but we should keep it in mind. Especially when it comes to handling qcodes modules with config entries.