microsoft/Qcodes #1158: Request for complex numbers in database

ThorvaldLarsen, 2018-06-25:
More and more data measured is inherently complex numbers so would like to request that the database can handle them directly. After our discussion last week in Delft it is clear that both sites would benefit from this capability.

I thought we can use this issue to discuss the whether this is a possibility or not.

@core @wpfff @nataliejpg

jenshnielsen, 2018-06-25:
Of relevance http://sqlite.1065341.n5.nabble.com/Thoughts-on-storing-arrays-of-complex-numbers-td81761.html

ThorvaldLarsen, 2018-06-28:
Interesting read. I know next to nothing about database storage so cant really comment on the implementation or how complicated it is.
However, it does seem to me that qcodes should handle complex numbers rather than everyone coming up with separate ad hoc solutions to store/open two real numbers instead of a single complex number.

astafan8, 2019-06-21:
Following PRs address this:
https://github.com/QCoDeS/Qcodes/pull/1592
https://github.com/QCoDeS/Qcodes/pull/1568
https://github.com/QCoDeS/Qcodes/pull/1493
https://github.com/QCoDeS/Qcodes/pull/1489
https://github.com/QCoDeS/Qcodes/pull/1484

closed 2019-06-21 by astafan8