microsoft/Qcodes #1174: Parameter sets value when value is out of range

lucblom, 2018-07-05:
### Steps to reproduce
```
from qcodes import Parameter
from qcodes.utils.validators import Numbers

the_value = -10

def set_function(value):
    global the_value
    the_value = value
    print('set {0}'.format(the_value))

def get_function():
    print('get {0}'.format(the_value))
    return the_value

a = Parameter('test', set_cmd=set_function, get_cmd=get_function, vals=Numbers(0, 100), step=5)
a.set(10)
a.get()
```

### Expected behaviour
```
a.set(10) --> ValueError('-10 is invalid')
a.get() --> -10
```
### Actual behavior
```
a.set(10) --> ValueError('-5 is invalid')
a.get() --> -5
```
### System
**operating system**
Windows 10 x64, miniconda v 4.5.4 (venv)

**qcodes branch**
master

**qcodes commit**
8970a083c2b582e4d966da082f039fef49f2089a

@dpfranke, @eendebakpt

jenshnielsen, 2018-07-05:
The problem is that the wrapper validated the end point but does not take into account that the points in between generated by step may not be valid because the parameter is out of bounds for the validator before starting to step.

jenshnielsen, 2018-07-05:
I believe that this is fixed by #1176. @lucblom Do you have any other ways of triggering this that does not involve both starting from an invalid value and setting a step size

jenshnielsen, 2018-07-05:
And by the way I disagree slightly on the expected behavior. I would expect it to be

```
a.get() --> -10
a.set(10) --> a.set(10) --> ValueError('-5 is invalid')
a.get() --> -5
a.get() --> -10
```

There is nothing in the code that will atempt to set the value to -10 so you should get a ValueError when trying to set the first step which is -5 but before the actual step is performed

lucblom, 2018-07-06:
@jenshnielsen. Thank you for the quick reply! 

We stumbled upon this behavior when using the IVVI. We use validators with a smaller range then the actual voltage limits of the device. We want to protect the user from setting to large values that could potentially damage the connected chip. How the invalid value was initially set is unclear to us. 
 
I only could trigger this when starting from an invalid value and I agree that the expected behavior should be as you described.