microsoft/Qcodes #1194: Sometimes database gets locked if a running measurement is interrupted

astafan8, 2018-07-17:
This is not easy to reproduce. So, he is what happens: a measurement is running in a Jupyter notebook, data is being saved, then the measurement gets interrupted (say KeyboardInterrupt), and then the .db file gets locked. This means that no other program (including the running python kernel) can access the database, the returned error message states that the database is locked.

Restarting the python kernel solves this problem, meaning it unlocks the database.

I have a feeling that the low-level code that handles interaction with sqlite database does not always unlock the database in case of an exception.

WilliamHPNielsen, 2018-09-07:
I think this has now been seen at a QT fridge in Copenhagen. They do, however, also perform some reading in separate processes. I'll try to look into this a bit.

WilliamHPNielsen, 2018-09-07:
This may or may not be relevant:
http://beets.io/blog/sqlite-nightmare.html

tl;dr
Implement atomic transactions the way it's done in #1002 to avoid `database is locked` errors.

nikhartman, 2019-07-08:
Just dropping a note that this happened to me and my issues search brought me here. I'm on `670004b06`.

astafan8, 2019-07-08:
@nikhartman Could you describe what happened in your case? What processes were running? Perhaps you have a full exception message?

nikhartman, 2019-07-09:
@astafan8 Sorry to say, I did not copy the traceback. I was running something along the lines of...

```python
for i in range(10):
    # setup measurement
    try:
        do1d(.......)
    except KeyboardInterrupt:
        pass
  
    # cleanup
```
I caught a KeyboardInterrupt (sent from the Jupyter dropdown menu) and got an exception that there was a database error (again, sorry I should have grabbed that). Subsequent measurements failed due to the database being locked.

colinDietrich, 2025-05-13:
Hello, I recently encountered the same **"database is locked"** error and wanted to share what I found in case it can help!

**Context**
we are running monitoring experiments that we often stop manually with a keyboard interrupt. The issue seems to occur during long-running database writes, especially if multiple `KeyboardInterrupts` are sent (which, admittedly, is not best practice) while an atomic transaction is in progress.

**Steps to Reproduce**
Although it's not straightforward, I was able to reliably trigger the "database is locked" error with:
- QCoDeS version 0.52.0
- Adding a 2-second sleep in `qcodes/dataset/sqlite/query_helpers.insert_many_values` (inside the `atomic` context manager, outside the `for` loop) to "simulate" a long transaction
- Running a script similar to the attached [file](https://github.com/user-attachments/files/20187881/loked_db_error.txt)
- Sending multiple `KeyboardInterrupts` during a transaction (happens every 5 seconds by default), then rerunning the experiment in the same IPython terminal, which results in the "database is locked" error

**Root Cause (as I understand it)**
- Data is stored during measurement and written to the database periodically or on exit via `DataSaver`.
- Database writes use the `atomic` context manager in `qcodes/dataset/sqlite/connection.py`, which ensures all operations are part of a single transaction.
- The `atomic` function uses a custom `DelayedKeyboardInterrupt` context manager to delay SIGINTs, but a second SIGINT triggers an immediate `KeyboardInterrupt`.
- If this happens, the `atomic` function does not catch the `KeyboardInterrupt`, so the transaction is not rolled back.
- The `KeyboardInterrupt` is then caught higher up (in the measurement context manager), which tries to access the database and flush data, but the previous transaction is still open, leading to further errors (e.g., AssertionError).
- The database connection is not properly closed, leaving the database locked for subsequent operations.

**Quick Fix**
A quick fix could be to ensure transactions are rolled back in `qcodes.dataset.sqlite.connection.atomic` if a KeyboardInterrupt occurs, before reaching  the measurement context manager. For example:
```python
except KeyboardInterrupt:
    conn.rollback()
    log.exception("KeyboardInterrupt caught! Rolling back transaction.")
    raise
```

In our case, catching any exception occurring when running those monitoring measurements to make sure connection to the database are rolled back and closed temporarily solve the issue.

Hope this helps!

mvanhorn, 2026-10-06:
atomic only rolled the connection back in an except Exception handler. KeyboardInterrupt is a BaseException, so an interrupt inside the transaction skipped that handler, never called rollback, and left the connection inside the open transaction that holds the database lock.

atomic now catches KeyboardInterrupt, calls conn.rollback(), logs "Rolling back due to unhandled exception", and re-raises. The docstring notes that a KeyboardInterrupt rolls the transaction back and is re-raised.

closed 2026-10-07 by jenshnielsen