microsoft/Qcodes #1208: driver / Agilent E8267C changes state on reinitialisation

WilliamHPNielsen, 2018-07-26:
See here:
https://github.com/QCoDeS/Qcodes/blob/f7b173fa88afae1248a732aeb2051a8e1c169b3c/qcodes/instrument_drivers/agilent/E8267C.py#L80-L81

As far as I understand, this is against our philosophy.

We do **not** want the `__init__` of a driver to change the state of the instrument.

astafan8, 2018-07-26:
I agree with the philosophy. 

So, for HP, i think we can get rid of it.

But for agilent, it looks like a safety thing (to set everything to 0).

WilliamHPNielsen, 2018-07-26:
Safety... -it's just a signal generator. I'd say it's an arbitrary choice, just like an amplitude of `0.1`

https://www.keysight.com/en/pd-57267-pn-E8267C/psg-vector-signal-generator?cc=US&lc=eng