microsoft/Qcodes #1247: Missing order of setpoints in the new dataset

Dominik-Vogel, 2018-08-27:
Currently the retrieval of setpoint data relies on the order of the registered dependent paramenters. So for example the code below will fail for `depends_on_wrong` and pass for `depends_on_right`

```
import qcodes as qc
from qcodes import ParamSpec, new_data_set, new_experiment
from qcodes.dataset.database import initialise_database
from qcodes.dataset.data_export import get_data_by_id

initialise_database()
new_experiment('issue1239', 'test2')

indep1 = ParamSpec('indep1', "numeric")
indep2 = ParamSpec('indep2', "numeric")
depends_on_right = [indep1, indep2]
depends_on_wrong = [indep2, indep1]
dep = ParamSpec('response', "numeric", depends_on=depends_on_wrong)

dataset = new_data_set("results", specs=[indep1, indep2, dep])
dataset.add_result({'response': 1234,
                    'indep2': 2,
                    'indep1': 1 })
dataset.mark_complete()

d = get_data_by_id(dataset.run_id)
# test if data is correctly saved
for el in d[0]:
    if el['name'] == 'indep2':
        val = el['data']
        break
# val[0] is actually 1, i.e indep2 and indep1 are interchanged
assert val[0] == 2
```

In the failing case the values for the setpoint axes get interchanged, because the array of dependent ParamSpecs is reversed. 

This behavior is of course undesired and has to be resolved. 
In my opinion this problem roots all the way back to the `get_setpoints` method of `sqlite_base.py`, which returns the setpoints as a `List[List[List[Any]]]`, for a parameter specified by its name. I don't think that we should establish some standard order of the setpoints to resolve this problem. We should possibly rather return a dictionary with the setpoint parameter names as keys instead of one level of the lists.

I will start identifying the uses of the named function and thereafter start implementing a fix that involves a dictionary for this purpose.

astafan8, 2018-08-27:
fully agree. Beware though that @WilliamHPNielsen has started working on a new format for storing dependencies information in relation to the data model. Perhaps, you can align with him.

WilliamHPNielsen, 2018-08-28:
@astafan8, good point. See https://github.com/QCoDeS/Qcodes/pull/1227 for a description of what is about to change. Unfortunately, https://github.com/QCoDeS/Qcodes/pull/1227 (preferably) requires https://github.com/QCoDeS/Qcodes/pull/1002 to be merged.

Dominik-Vogel, 2018-08-28:
Thanks for pointing that out. It is good that we are all aware of those correlations. It seems to me that #1227 will require a reimplementation of `get_data_by_id` . This reimplementation should include the changes proposed here. I think implementing these changes already in the current version will not impose extra work.

PR #1250 (2018-09-04): Fix  requirements of ordering of setpoints as dependencies
Fixes #1247
closed 2018-09-04 by Dominik-Vogel