microsoft/Qcodes #1275: Move data type checking against paramtype of ParamSpec into DataSet

astafan8, 2018-09-18:
It seems more logical that `DataSet`'s register_parameter method takes care of type validation of the data against its `ParamSpec`, because `ParamSpec` is on the same level as `DataSet`. Currently, the logic is in `DataSaver`, hence its scope is now to large. `DataSaver` should only know how to interface between qcodes parameters (there are many different classes) and `DataSet`, but not go deep into type checking, etc.

WilliamHPNielsen, 2018-09-18:
While I agree that there is too much going on in the `DataSaver` (we are working on refactoring some of that into `InterDependencies` in #1227), I can't really see any alternative when it comes to validating the type of the *data as it is being saved*. Am I missing your point?

astafan8, 2018-09-19:
You're right, but the data is *actually* saved in `DataSet`, not in `DataSaver`. `DataSaver`'s role is to interface between qcodes parameters world and `DataSet`, right? Hence `DataSaver` just unwraps Parameters, ArrayParameters, etc. and passes values to `DataSet`. And since only `DataSet` knows about `ParamSpec`s of the parameters, the type checking of values against types in `ParamSpec` belongs to `DataSet`. This is my point.

WilliamHPNielsen, 2018-09-19:
Thanks for clarifying. So although the datasaver must command _when_ the validation is done, the code that knows _how_ to validate (and thus actually performs the validation) belongs somewhere else. Is that it? If so, wouldn't it then seem appropriate to make the `ParamSpec`s have associated validators? (I'm not saying that the `ParamSpec`s are parameters, but they are parameters)

astafan8, 2019-06-20:
closing as outdated, and not really relevant anymore.

closed 2019-06-20 by astafan8