microsoft/Qcodes #1285: Adding metadata to the new dataset

nataliejpg, 2018-09-28:
If I use the dataset as created by 

` dataset = qc. new_data_set('results', specs=[some_paramspec]) `

then I can add metadata to it like:

`dataset.add_metadata('measurement_notes', 'really warm today')`

but if I use the `Measurement` object to create a dataset like:

```
analysis_meas = Measurement()
analysis_meas.register_parameter(some_parameter)
with analysis_meas.run() as datasaver:
    datasaver.add_result((some_parameter.name, 5))
```
Then I don't see how I am meant to add metadata as neither `Measurement` nor `Datasaver` have an easy way to access the dataset or to add metadata. 

In a related note it would be great to have an example of adding metadata in the examples and similarly of retrieving it (minimally how to retrieve the snapshot should be there unless I'm missing something(?)).

astafan8, 2018-10-01:
Just to add some information about problems and *danger* of the `add_metadata` method:

> The option that you actually mentioned will not really work, unfortunately. Let me explain. Within the context manager, you can access the `DataSet` object via the `datasaver` object as `datasaver.dataset` (e.g. using "dataset" attribute of "datasaver"), and then you can call `add_metadata` method. However, this method is actually quite dangerous (for example, see https://github.com/QCoDeS/Qcodes/issues/898). So my suggestion would be to use it like `dataset.add_metadata('snapshot', <your data regardless of its format/representation>)` (note that for snapshots we use JSON, so I recommend you to also convert your metadata to JSON for consistency/convenience). The problem with this usage is that it will *overwrite* (!) the snapshot data of the station, which is absolutely not handy. So, ideally, first, the snapshot needs to be retrieved `dataset.get_metadata('snapshot')`, then converted to Python dict, then an extra metadata can be added, the result converted to JSON, and only then the complete thing should be written again via `dataset.add_metadata('snapshot', <JSON representation of the whole metadata>)`. This is very ugly and cumbersome and error prone, but it is a recipe that you can use know *if it is absolutely necessary*.

mgunyho, 2019-09-26:
Has there been any work on this? I would very much like to add metadata to my runs.

astafan8, 2019-09-26:
no yet. this hasn't got enough prio yet. But contributions are welcome.

mgunyho, 2019-09-27:
Alright, I can try to look into this at some point, although I'm not that familiar with SQL.