microsoft/Qcodes #1315: Standard way to retrieve data fails if no dependencies exist

WilliamHPNielsen, 2018-10-15:
The function `get_data_by_id`, which is also being used by `get_shaped_data_by_runid`, only looks for dependent variables. That means that a flat/dependency-less run can not have its data retrieved by this function. Is this really what we want?

Here is a small example:
```python
import numpy as np

from qcodes.dataset.data_export import get_data_by_id
from qcodes.dataset.measurements import Measurement
from qcodes import Parameter

x = Parameter('x')
y = Parameter('y')

meas = Measurement()
meas.register_parameter(x)
meas.register_parameter(y)  # no data retrieved
# meas.register_parameter(y, setpoints=(x,))  # yes data retrieved

with meas.run() as datasaver:

    for x_val, y_val in enumerate(np.random.randn(25)):

        datasaver.add_result((x, x_val), (y, y_val))

data = get_data_by_id(datasaver.run_id)
print(data)
```

astafan8, 2018-10-15:
I guess that `get_data_by_id` was designed only for those cases where dependent parameters exist. The structure of what it returns indeed implies that there are dependent parameters.

We should be careful with the "is it really what we want" question. The answer depends on the purpose of `get_data_by_id`. If `get_data_by_id` is __the__ main/default way to extract data from the sqlite-backended dataset, then i'd agree that we want this function to properly work in the case where there are no dependent parameters. If `get_data_by_id` is just one of the ways to retrieve data, and there is a different default/way, then I'd say we just should document this fact in the function's docstring (and add test for it as well).

astafan8, 2019-12-06:
yes, "`get_data_by_id` is just one of the ways to retrieve data", and is discouraged and within qcodes used only in `plot_dataset`, anyway `DataSet.get_parameter_data` is the preferred method.

@WilliamHPNielsen shall we close this as "not relevant anymore"?