microsoft/Qcodes #1335: Driver / Infiniium oscilloscope : feature requests

WilliamHPNielsen, 2018-10-24:
As requested by our friends in Sydney.

They would love:

1. To have a way to use the streaming method for large number of points as it is possible to hit the 1GB limit.
2. For the get trace to return the current trace not to trigger and wait. (don't use DIGITIZE)
3. The sin(x)/x interpolation to be disabled so number of points requested = the number of points returned
4. To have functions to control the state of the scope ie Run, Single, Stop
5. A function to block until the scope has been triggered and acquired its data/number of averages. using ESR/STB polling or equivalent instead of *OPC? and a long timeout as the scope becomes unresponsive if we decide to interrupt the program.
6. For the `trace` command not to change the acquisition mode to real-time. It shouldn't be reset every time `get_raw()` is called, since the scope may be in equivalent-time or high-res mode

@StevenWaddy @Dominik-Vogel

WilliamHPNielsen, 2018-10-24:
In doing this, me may want to keep #1147 in mind.

StevenWaddy, 2018-10-24:
> Seems like a very good and understandable list of requests to me.
> For the realization it seems to me that we can support both the streaming mode and the shot-by-shot mode. This means for 2.) that we have to split the current get function up into one that acquires data and one that transfers the data. (So actually two to acquire data when including the streaming mode).
> 3.) is already implemented

Streaming mode will work for all lengths of data so is there a reason to keep the non-streaming mode?
Its also worth noting that splitting the acquisition and the transfer is the only way to read multiple traces captured at the same moment in time. So I may call the acquire command, then call the transfer command for each channel I care about.

Also with issue #1147<https://protect-au.mimecast.com/s/sig2CROAQotAolXysNkg6w?domain=github.com>. It wasn't present in the original driver / the derivatives in use in Sydney. The only differences I can see are the Sydney driver uses streaming mode and I also manipulated the raw data by hand so it may be an issue with non-streaming or the pyvisa binary query command.

neildick, 2019-01-29:
Sorry to bump this, but I'm about to start some new measurements using this scope. Did you guys have a chance to look at some of these features?

commit 1e0e9385 (2022-02-17): Merge #3916

3916: Feat: Improved Keysight Infiniium Driver r=jenshnielsen a=spauka

This PR adds a new driver for the Keysight Infiniium series of oscilloscopes. It solves several outstanding issues with the previous driver and adds a number of new useful features.

Closes #1147 (uses streaming mode) and closes #1335

~Depends on #3889 since we use the := operator.~ removed this since py37 might be sticking around :D

## Changes and Features
 - Trace parameters are updated to use `ParameterWithSetpoints`.
 - The driver now works "out of the box", removing the need to `prepare_curvedata` befor [... cut]
PR #3916 (2022-02-17): Feat: Improved Keysight Infiniium Driver
This PR adds a new driver for the Keysight Infiniium series of oscilloscopes. It solves several outstanding issues with the previous driver and adds a number of new useful features.

Closes #1147 (uses streaming mode) and closes #1335

~Depends on #3889 since we use the := operator.~ removed this since py37 might be sticking around :D

## Changes and Features
 - Trace parameters are updated to use `ParameterWithSetpoints`.
 - The driver now works "out of the box", removing the need to `prepare_curvedata` before measurement. For example, the following co [... cut]
closed 2022-02-17 by bors[bot]