microsoft/Qcodes #1347: QtPlot with sweeps

patomaki, 2018-10-28:
Hi! 
Thanks for the good work on the environment/library. 

I was trying to enable dynamic 2d-map plotting with QtPlot. What I need is a way to add traces one by one, i.e. sweep by sweep (which appears to be a hassle in every aspects with QCoDeS). What I tried so far is 
1. extract x_data, y_data, z_data as qc.DataArrays of sizes (n,) (1,), and (n,), respectively
2. after the first trace, create plot2d = qc.QtPlot(x_data, y_data, z_data)
3. add subsequent traces with the add method, as plot2d.add(x_data_new, y_data_new, z_data_new) (here,  x_data_new, y_data_new, z_data_new are qc.DataArrays of sizes (n,) (1,), and (n,) as before)
4. Then call plot2d.autorange() to keep to the original figure size.

What I want: one 2d-map plot with sweeps added one by one, position based on their y-value. One x-axis, one y-axis, one z-axis and associated colorbar get their ranges updated. (What else I want: exactly one DataSet created over the course of all the sweeps.)

The above otherwise works, but has the unexpected feature that each add-method also adds a separate colorbar, which is inconvenient if there are going to be 500 traces. I found no ways to control the colorbars (it's not done via update_colorbar = True in autorange). I failed to interface with the likes of plot.update() to get what I want, probably because I fail to employ loops to get what I want*.

*Also, any suggestions on how to natively use DataSets which allow adding data sweep by sweep would be greatly appreciated. The reason I need this is that the sweep ranges and sweep directions might change at every sweep. I also might address each trace separately later.

I'm working on an Mac, and installed qcodes a week or two ago on the machine. 

Best wishes, and thanks for your time!

WilliamHPNielsen, 2018-10-30:
Hi @patomaki! Welcome to the discussion.

Let's jump right in.

> [doing what I want] appears to be a hassle in every aspects with QCoDeS

You have discovered one of the reasons why we made the new dataset and want to abandon the `Loop` altogether. I am fairly sure that what you are trying to achieve _measurement-wise_ is very easy with the `Measurement` object. For plotting it, you could  have a look a plottr. https://github.com/kouwenhovenlab/plottr/blob/master/plottr/plottr.py

patomaki, 2018-10-30:
Dear @WilliamHPNielsen ,

Thanks for the help! I just took a look at plottr. Alas, plottr documentation explicitly states that e.g. zig-zag sweeps, or other more complex sweep patterns are not supported ( https://github.com/kouwenhovenlab/plottr/blob/master/doc/live_plotting_for_QCoDeS.ipynb ), so it wouldn't work for me. 
Additionally, if I was to use the new data set class, is there a qcodes-way to convert the output from the database to a regular folder-.dat file-.png-based output? (I can always do this myself)

Best wishes!

astafan8, 2018-10-30:
@patomaki beware that `plottr` is being updates sometimes, but (unfortunately) not necessarily the notebooks. Hence the not-working cases might already work :)

regarding "convert the output from the database to a regular folder-.dat file-.png-based output": such conveniences do not yet exist in qcodes. However, there are functions that you can use to extract data and then save it in the format that you like. We welcome PRs which implement this as we do not seem to have a bandwidth at the moment. So, feel free to contribute!

WilliamHPNielsen, 2018-10-31:
@patomaki, sorry that plottr was perhaps a dead end. As a fallback, you can *always* use `plot_by_id` to plot *any* dataset (of the new type). Would you be able to use that? Or is the live-update essential here?

WilliamHPNielsen, 2018-11-05:
@patomaki Have you joined the QCoDeS slack channel? I think I can help you with a quick workaround solution, but we'd have to discuss a bit in "real time". 

To join slack, email one of us: http://qcodes.github.io/Qcodes/community/contributing.html
If you're already there, feel free to PM me.