microsoft/Qcodes #1369: test_subscription_from_config is flaky

jenshnielsen, 2018-11-05:
I.e here https://travis-ci.org/QCoDeS/Qcodes/jobs/450752810 in #1366

```
________________________ test_subscription_from_config _________________________
dataset = test-dataset #1@/tmp/tmplf98kwxq/temp.db
----------------------------------------
x - numeric
y - numeric
basic_subscriber = <function basic_subscriber.<locals>.subscriber at 0x7f74ec0bc400>
    def test_subscription_from_config(dataset, basic_subscriber):
        """
        This test is similar to `test_basic_subscription`, with the only
        difference that another subscriber from a config file is added.
        """
        # This string represents the config file in the home directory:
        config = """
        {
            "subscription":{
                "subscribers":{
                    "test_subscriber":{
                        "factory": "qcodes.tests.dataset.test_subscribing.MockSubscriber"
                        "factory_kwargs":{
                            "lg": false
                        }
                        "subscription_kwargs":{
                            "min_wait": 0
                            "min_count": 1
                            "callback_kwargs": {}
                        }
                    }
                }
            }
        }
        """
        # This little dance around the db_location is due to the fact that the
        # dataset fixture creates a dataset in a db in a temporary directory.
        # Therefore we need to 'backup' the path to the db when using the
        # default configuration.
        db_location = qcodes.config.core.db_location
        with default_config(user_config=config):
            qcodes.config.core.db_location = db_location
    
            assert 'test_subscriber' in qcodes.config.subscription.subscribers
    
            xparam = ParamSpec(name='x', paramtype='numeric', label='x parameter'
                               unit='V')
            yparam = ParamSpec(name='y', paramtype='numeric', label='y parameter'
                               unit='Hz', depends_on=[xparam])
            dataset.add_parameter(xparam)
            dataset.add_parameter(yparam)
    
            sub_id = dataset.subscribe(basic_subscriber, min_wait=0, min_count=1
                                       state={})
            sub_id_c = dataset.subscribe_from_config('test_subscriber')
            assert len(dataset.subscribers) == 2
            assert list(dataset.subscribers.keys()) == [sub_id, sub_id_c]
    
            expected_state = {}
    
            for x in range(10):
                y = -x**2
                dataset.add_result({'x': x, 'y': y})
                expected_state[x+1] = [(x, y)]
>               assert dataset.subscribers[sub_id].state == expected_state
E               assert {1: [(0, 0)]} == {1: [(0, 0)]}
E                 Omitting 1 identical items, use -vv to show
E                 Use -v to get the full diff
tests/dataset/test_subscribing.py:142: AssertionError
```

astafan8, 2018-11-05:
Is it smth that can be solved in the same (ugly) fashion as [this](https://github.com/QCoDeS/Qcodes/blob/68f45cdd5c50933bc4a3f2d31f830811c68ad5b6/qcodes/tests/dataset/test_measurement_context_manager.py#L485) ?

Dominik-Vogel, 2018-11-05:
The problem is probably due to multithreading. I think we should not spend a lot of time on this as we are going to redesign the subscription process anyways.... I will try to add a short wait and reduce the number of repetitions.

PR #1372 (2018-11-05): Fix flaky `test_basic_subscription` 
fixes #1369 (possibly)

TIL: force push to a closed PR will render it unreopenable.
closed 2018-11-05 by Dominik-Vogel