microsoft/Qcodes #1424: Tests fail due to missing websockets requirement

StefanD986, 2018-12-21:
While setting up Qcodes for a colleague we ran into the issue that running the unit tests locally fails with ModuleNotFoundError (`websockets`).

This was probably introduced as part of #1407 . The fix is easy, by adding websockets to the test_requirements.txt file. 

But the real issue I think is that there are several places where dependencies are defined (setup.py, requirements.txt, test_requirements.txt, docs_requirements.txt). I am wondering why the CI didn't catch that? does it execute the unit tests in the same env as where it install Qcodes?

If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. Clone latest Qcodes
2. Create virtualenv, and `pip install -r requirements.txt -r test_requirements.txt -r docs_requirements.txt`
3. run pytest in qcodes/tests

### Expected behaviour
Tests should all pass

### Actual behaviour
Tests throw:

```Python
qcodes/tests/test_monitor.py:None (qcodes/tests/test_monitor.py)
ImportError while importing test module '/Users/swdroge/PycharmProjects/Qcodes/qcodes/tests/test_monitor.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test_monitor.py:9: in <module>
    import websockets
E   ModuleNotFoundError: No module named 'websockets'
```

### System
**operating system**: OSX & Windows

**qcodes branch**: master

**qcodes commit**: 023d4189

WilliamHPNielsen, 2018-12-27:
@StefanD986, you should also install QCoDeS itself in order to be able to run the tests.

WilliamHPNielsen, 2018-12-27:
For reference, this is what Travis does:
https://github.com/QCoDeS/Qcodes/blob/023d4189fbe762d6532c18b95447171a5ab3569f/.travis.yml#L25-L30

StefanD986, 2018-12-27:
Looks like a lot of steps and duplication to run some tests.

As far as I can see it the current state is:
  - `setup.py`: runtime requirements
  - `requirements.txt`: runtime requirements (but not all)
  - `test_requirements.txt`: development requirements such as test frameworks
  - `docs_requirements.txt`: packages needed to build docs, plus `jupyter` to execute examples, and support packages for some of the notebooks (`pandas`)

May I suggest the following:
  - Let's use `setup.py` for runtime dependencies only
  - let's use `requirements.txt` for development dependencies only.
    - As development dependencies are a superset of runtime dependencies, let's include setup.py automatically by adding `.` to `requirements.txt`.
     - additionally add test framework / utility packages needed to test the code (so mostly the stuff that now is in `test_requirements.txt`)

I am indifferent about keeping `docs_requirements.txt` separate. Doesn't hurt for me to keep them separate, especially since `jupyter` is a bit of a heavy-weight dependency.

the proposed changes would condense the `install` in `travis.yml` to:

```yaml
install:
    - pip install --upgrade pip 
    - pip install -r requirements.txt 
    - pip install -r docs_requirements.txt 
```

I'll create a PR with the proposed changes.

astafan8, 2019-06-20:
Closing because #1425 has been closed. Also, websockets are used for qcodes Monitor which we have tests for, so there should be no issue (see William's comment).

closed 2019-06-20 by astafan8