microsoft/Qcodes #1439: live plotting with Measure.run()

nikhartman, 2019-01-11:
New here, quick question:

It seems from the discussion here that the `Loop` is meant to be depreciated in favor of looping with a `Measure.run()` object. Coming from the Folk lab, we're still running Igor code that looks a lot like your `doNd()` examples.

What would be the preferred way to add a live plot to that example? It looks like maybe the answer is `plottr`, which is fairly clear how to implement, but not super clear which repository is under active development.

astafan8, 2019-01-11:
You are almost correct, the preferred way is to use `Measurement` (not to be confused with `Measure`), [here is the doc about it](http://qcodes.github.io/Qcodes/examples/DataSet/Dataset%20Context%20Manager.html).

For live plotting, you can indeed use `plottr`. At the moment, active development of `plottr` happens on github.com/kouwenhovenlab/plottr.

nikhartman, 2019-01-11:
Thanks for the reply. That looks good.

To be a little more general, what is intended is for live plotting to happen through a subscriber, rather than using the `MatPlot` and `QtPlot` classes? Is there any reason not to use those classes and subscribe a callback that runs their `.update()` methods?

nikhartman, 2019-01-11:
> You are almost correct, the preferred way is to use `Measurement` (not to be confused with `Measure`), [here is the doc about it](http://qcodes.github.io/Qcodes/examples/DataSet/Dataset%20Context%20Manager.html).

That is definitely what I meant, but I just caught on to why that distinction is so important. Changed title for accuracy.

astafan8, 2019-01-14:
> what is intended is for live plotting to happen through a subscriber, rather than using the `MatPlot` and `QtPlot` classes? Is there any reason not to use those classes and subscribe a callback that runs their `.update()` methods?

I don't think that this combination has been tested. I personally have not looked into if they are compatible. Those classes are legacy ones, and most probably have limitations that are similar to those of `Loop`. On the other hand, if you are interested in trying it, and it turns out that that combination works well for your experiment, I don't see any reason of not using it (apart from the "legacy" notions) :)

nikhartman, 2019-01-14:
I'll tinker with the various options and post back about any success. Thanks for pointing me in the right direction.

closed 2019-01-14 by nikhartman