microsoft/Qcodes #1457: default subscriber for plottr does not work for its new version

astafan8, 2019-01-28:
For implementing live plotting, the new `plottr` monitors the database file directly, without using the subscribers, hence the information in [qcodes docs](http://qcodes.github.io/Qcodes/examples/plotting/live_plotting.html#Live-plotting-with-plottr-as-an-example) is not correct anymore.

See https://github.com/kouwenhovenlab/plottr/issues/36.

I suggest just to remove the mentioning of `plottr`, or substitute it with a link the exactly working version (which is not ["the old `plottr`"](https://github.com/kouwenhovenlab/plottr/releases/tag/original-final))

Dominik-Vogel, 2019-02-05:
I agree - go for it!
Monitoring the db file is really all it takes.

jenshnielsen, 2023-02-09:
Plottr no longer uses subscriptions

closed 2023-02-09 by jenshnielsen