microsoft/Qcodes #1600: Dataset.get_data() sometimes does not return the correct row(s) of data

thibaudruelle, 2019-06-18:
### Steps to reproduce

```
qc.new_experiment(name='get_data_issue', sample_name='none')

class MockMultiParam(MultiParameter):
    def __init__(self):
        super().__init__("multiparam", names=('A','B'), shapes=((10,),)*2, setpoints = ((range(10),),)*2, setpoint_names = (('t',),)*2 )
        self.i = 0
        
    def get_raw(self):
        val = (np.arange(10) + self.i*10,)*2
        self.i +=1
        return val

mock_param = MockMultiParam()

meas = Measurement()
meas.register_parameter(mock_param, paramtype='array')
with meas.run() as datasaver:  
    datasaver.add_result((mock_param, mock_param()))
    run_id = datasaver.run_id

ds = load_by_id(run_id)

print(ds.get_data('B'))
print(ds.get_parameter_data('B'))

print(ds.get_data('B', start=1,end=1))
print(ds.get_parameter_data('B', start=1,end=1))

print(np.all(ds.get_data('B', start=1,end=1) == ds.get_parameter_data('B', start=1,end=1)['B']['B'][0]))
print(np.all(ds.get_data('B', start=2,end=2) == ds.get_parameter_data('B', start=1,end=1)['B']['B'][0]))
```

### Expected behaviour
`np.all(ds.get_data('B', start=1,end=1) == ds.get_parameter_data('B', start=1,end=1)['B']['B'][0])` should return `True`

### Actual behaviour
`np.all(ds.get_data('B', start=1,end=1) == ds.get_parameter_data('B', start=1,end=1)['B']['B'][0])`  returns `False`, `np.all(ds.get_data('B', start=2,end=2) == ds.get_parameter_data('B', start=1,end=1)['B']['B'][0])` returns `True`.

### System
**operating system**
Windows 10

**qcodes branch**
qcodes/master

**qcodes commit**
1fda5c5a345da6cd2831773054924a2382420020

jenshnielsen, 2019-06-19:
For reference the output of the print commands is. 
```
[[array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])], [None]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
[[array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
```

@thibaudruelle I took the liberty of modifying your example slightly so that is can be executed without any other imports. It it not trivially obvious that the output of get_data should be equivalent to the output of get_parameter_data. I assume that the issue are the additional None columns?

thibaudruelle, 2019-06-19:
@jenshnielsen I have a different print output running the example:
```
Starting experimental run with id: 129
[[None], [array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
[[None]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
False
True
```
I agree get_parameter_data and get_data should not necessarily return the same output. What I find misleading is that the first value of 'B' being `array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])`, `ds.get_data('B', start=1,end=1)` returns `None`, which seems wrong to me. 
I went into the code and it looks like the sql query in get_parameter_data as an extra `"WHERE {param} IS NOT NULL"` which I believe might fix `get_data`. 

Note that this issue was not originally present, the change in the behavior of `get_data` was likely caused by a change in how the data in saved in the database. Using commit 68d12462435f840cfef02c8fdb18773040a1f797, the same code returns:
```
Starting experimental run with id: 130
[[array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
[[array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])]]
{'B': {'B': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]]), 't': array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9]])}}
True
False
```

jenshnielsen, 2020-10-23:
`get_data` has been replaced by `get_paramter_data` which does not have these issues

closed 2020-10-23 by jenshnielsen