microsoft/Qcodes #1601: [feature request] get_parameter_data option to not return setpoints

thibaudruelle, 2019-06-18:
In the case of a `Multiparameter` with many internal setpoints that stay constant while an external setpoint is swept, the output of `Dataset.get_parameter_data` can get unwieldy. A useful option would be to only return setpoints once, then something similar to `Dataset.get_data`.

```
qc.new_experiment(name='get_data_issue', sample_name='none')

class MockMultiParam(MultiParameter):
    def __init__(self):
        super().__init__("multiparam", names=('A','B'), shapes=((1e6,),)*2, setpoints = ((range(1e6),),)*2, setpoint_names = (('t',),)*2 )
        self.i = 0
        
    def get_raw(self):
        val = (np.arange(1e6) + self.i*10*1e6,)*2
        self.i +=1
        return val

mock_param = MockMultiParam()
t2_param = Parameter('t2', set_cmd=None, get_cmd=None)

meas = Measurement()
meas.register_parameter(t2_param)
meas.register_parameter(mock_param, paramtype='array', setpoints=(external_param,))

with meas.run() as datasaver:  
    for t2 in range(100):
        datasaver.add_result((t2_param, t2), (mock_param, mock_param()))
        run_id = datasaver.run_id

print(ds.get_data('B'))
print(ds.get_parameter_data('B'))
```

jenshnielsen, 2019-06-19:
@thibaudruelle I had to modify your example a bit to get it to run, can you please confirm if the below is what you intend

* shape should be an int tuple not float. 
* arange does not take float so replace it with numpy arange
* external_param does not exist so replace it with t2_param. 
* ds is undefined. Replace it with an actual reference to the dataset

```python
import qcodes as qc
from qcodes import MultiParameter, Measurement, Parameter
from qcodes.dataset.data_set import load_by_id
import numpy as np

qc.new_experiment(name='get_data_issue', sample_name='none')

class MockMultiParam(MultiParameter):
    def __init__(self):
        super().__init__("multiparam", names=('A','B'), shapes=((1_000_000,),)*2, setpoints = ((np.arange(1e6),),)*2, setpoint_names = (('t',),)*2 )
        self.i = 0
        
    def get_raw(self):
        val = (np.arange(1e6) + self.i*10*1e6,)*2
        self.i +=1
        return val

mock_param = MockMultiParam()
t2_param = Parameter('t2', set_cmd=None, get_cmd=None)

meas = Measurement()
meas.register_parameter(t2_param)
meas.register_parameter(mock_param, paramtype='array', setpoints=(t2_param,))

with meas.run() as datasaver:  
    for t2 in range(100):
        datasaver.add_result((t2_param, t2), (mock_param, mock_param()))
        run_id = datasaver.run_id

ds = datasaver.dataset
        
print(ds.get_data('B'))
print(ds.get_parameter_data('B'))
```

jenshnielsen, 2019-06-19:
This works exactly as I would expect. Can you elaborate on what you mean by `unwieldy` it has the same shape as the data so it's no more `unwieldy` than the data

thibaudruelle, 2019-06-19:
@jenshnielsen Sorry for the mistakes in the working example, I corrected them.
It does work as expected, however in the specific case where the values of 't' do not depend on 't2', `get_parameter_data` returns a lot of redundant data. For my use case it would be useful to have an option to return only the `top_level_parameter` data. It might be a niche case and not worth working on it though.