microsoft/Qcodes #1645: Bad black magic import of config

WilliamHPNielsen, 2019-07-29:
As discovered by the dynamic duo @jenshnielsen and `mypy` in #1640, the import of the config module and the later overriding assignemnt in the main `__init__.py` file is a bit creepy.

https://github.com/QCoDeS/Qcodes/blob/1d08d81adbb173e5f8bd486ee2a2fc3454641b1e/qcodes/__init__.py#L7-L14

Like, what is `qcodes.config` now? What should it be?

I propose that we resolve this by renaming (with an appropriate deprecation cycle) the `qcodes.config` module to `qcodes.configuration`. I think that will have the minimal impact on user workflows.

@QCoDeS/core what do you think?

jenshnielsen, 2019-07-29:
I agree, This will however need careful testing to ensure that the currently (by blackmagic working) logic does not break

astafan8, 2019-07-29:
i agree with the issue, and the proposed solution.

Akshita07, 2020-07-16:
This issue was solved by #1682

closed 2020-07-16 by Akshita07