microsoft/Qcodes #1674: AttributeError of PPMS_private_server

zznature, 2019-08-16:
I have edit private server of QD_DynaCool to work with PPMS. Also the instrument driver is obtained mainly just by changing the name "DynaCool" to "PPMS".  

However, when the server talks to the controlling software 'MultiVu', AttributeError emerges. I guess the 'GetChamberTemp', 'GetLastTempSetpoint', 'GetFieldSetpoints' commands do not work for PPMS.

How would the AttributeError of PPMS_private_server be solved? 

### Expected behaviour
The commands of PPMS is expected to be the same with DynaCool. Most of the commands work properly, such as GetTemperature,  GetField, *IDN. Right status of the PPMS could be obtained.

`'TEMP': CmdArgs(cmd=self._mvu.GetTemperature,
                                      args=[_variants['double'],
                                            _variants['long']]),`

### Actual behaviour
However, 3 commands listed below get AttributeError on the PPMS dedicated control PC when running the server.py.

`		     'CHAT': CmdArgs(cmd=self._mvu.GetChamberTemp,
                                      args=[_variants['double'],
                                            _variants['long']]),
                      'GLTS': CmdArgs(cmd=self._mvu.GetLastTempSetpoint,
                                      args=[_variants['double'],
                                            _variants['double'],
                                            _variants['long']]),
                      'GLFS': CmdArgs(cmd=self._mvu.GetFieldSetpoints,
                                      args=[_variants['double'],
                                            _variants['double'],
                                            _variants['long'],
                                            _variants['long']]),`
### System
**operating system**
Windows
**qcodes branch**
0.4.0
**qcodes commit**

WilliamHPNielsen, 2019-08-19:
Hi @zznature, thanks for reaching out. I must admit that I am not completely sure what you are doing. Could you elaborate on what you mean by
> I have edited the private server of QD_DynaCool to work with PPMS.
?
It might be good if you could refresh my memory on what the instruments are called and what they do. 

In any case, I agree that it looks like you are trying to call non-existing methods of the MultiVU API. Do you have any API documentation from the instrument vendor? When I wrote the DynaCool driver, I used the instrument simulator (proprietary software, can be obtained by writing to the instrument vendor and asking nicely) to look up the API calls and their call signatures. It was rather cumbersome, but no proper documentation existed.

zznature, 2019-08-19:
Thanks! @WilliamHPNielsen 
DynaCool is the cryo-free version of PPMS, with some updates possibly, both of which are made by QuantumDesign, Inc. I have asked for the python socket_server files from QuantumDesign, which are in python 2.7 and the same for PPMS and DynaCool.

I have only changed the name from 'DynaCool' to 'PPMS' in the private_server.py and instrument driver of DynaCool.py in Qcodes. So I am wondering how to obtain the API for the controlling software, MultiVu. With your reminder, I have sent an email to QD asking the API.

Actually at present I could work with the results in the snapshot above. The value of temperature and field could be acquired, only the setting parameters are not acquired.

closed 2019-09-23 by zznature

fishyvishy, 2024-11-26:
> Thanks! [@WilliamHPNielsen](https://github.com/WilliamHPNielsen) Sorry that I did not express myself clearly. DynaCool is the cryo-free version of PPMS, with some updates possibly, in my opinion. Both are made by QuantumDesign, Inc. I have asked for the python socket_server files from QuantumDesign, which are in python 2.7 and the same for PPMS and DynaCool.
> 
> I have only changed the name from 'DynaCool' to 'PPMS' in the private_server.py and instrument driver of DynaCool.py in Qcodes. So I am wondering how to obtain the API for the controlling software, MultiVu. With your reminder, I have sent an email to QD asking the API. The PPMS MultiVu version in our lab is 1.5.11.
> 
> Actually at present I could work with the results in the snapshot above. The value of temperature and field could be acquired, only the setting parameters are not acquired.

Hello @zznature, 

We are trying to connect to a dynacool PPMS as well. After opening the MultiVU application, running qcodes-dynacool-server on the PPMS dedicated control PPMS gives the error below. Did you experience a similar error or Is there another step we have to take on the MultiVU application? 

Could not connect to Multivu Application. Please make sure that the MultiVu Application is running.
Traceback (most recent call last):
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\win32com\client\dynamic.py", line 80, in _GetGoodDispatch
    IDispatch = pythoncom.connect(IDispatch)
pywintypes.com_error: (-2147221005, 'Invalid class string', None, None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\qcodes\instrument_drivers\QuantumDesign\DynaCoolPPMS\private\commandhandler.py", line 51, in __init__
    self._mvu = win32com.client.Dispatch(client_id)
                ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\win32com\client\__init__.py", line 114, in Dispatch
    dispatch, userName = dynamic._GetGoodDispatchAndUserName(dispatch, userName, clsctx)
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\win32com\client\dynamic.py", line 100, in _GetGoodDispatchAndUserName
    return (_GetGoodDispatch(IDispatch, clsctx), userName)
            ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\win32com\client\dynamic.py", line 82, in _GetGoodDispatch
    IDispatch = pythoncom.CoCreateInstance(
        IDispatch, None, clsctx, pythoncom.IID_IDispatch
    )
pywintypes.com_error: (-2147221005, 'Invalid class string', None, None)
Traceback (most recent call last):
  File "C:\Users\QDUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\win32com\client\dynamic.py", line 80, in _GetGoodDispatch
    IDispatch = pythoncom.connect(IDispatch)
pywintypes.com_error: (-2147221005, 'Invalid class string', None, None)