microsoft/Qcodes #1698: Key error when setting units on AMI430

simonzihlmann, 2019-09-06:
The AMI430 magnet power supply produces an error while setting the units (either field_units or ramp_rate_units). Below, an example is given for the field_units.

### Steps to reproduce
1. Initialize power supply:
`ix = AMI430("x2", address="192.168.0.101", port=7180)`
2. Read units from power supply (example given for field_units):
`ix.field_units()`
This produces the following expected output:
`'tesla'`
3. Set unit (example given for field_units):
`ix.field_units('kilogauss')`
This does not work as expected

4. The error can also be reproduced by:
`ix.field_units(ix.field_units())`

### Expected behaviour
It should silently set the field_unit to kilogauss (or tesla). Without any error.

### Actual behaviour
It produces a key error:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-7-f9d194d73bef> in <module>
----> 1 ix.field_units('kilogauss')

c:\users\g-gre-gre058050\qcodes\qcodes\instrument\parameter.py in __call__(self, *args, **kwargs)
    354         else:
    355             if hasattr(self, 'set'):
--> 356                 self.set(*args, **kwargs)
    357             else:
    358                 raise NotImplementedError('no set cmd found in' +

c:\users\g-gre-gre058050\qcodes\qcodes\instrument\parameter.py in set_wrapper(value, **kwargs)
    557             except Exception as e:
    558                 e.args = e.args + ('setting {} to {}'.format(self, value),)
--> 559                 raise e
    560 
    561         return set_wrapper

c:\users\g-gre-gre058050\qcodes\qcodes\instrument\parameter.py in set_wrapper(value, **kwargs)
    541                     t0 = time.perf_counter()
    542 
--> 543                     set_function(raw_value, **kwargs)
    544                     self.raw_value = raw_value
    545                     self._save_val(val_step,

c:\users\g-gre-gre058050\qcodes\qcodes\utils\command.py in __call__(self, *args)
    175             raise TypeError(
    176                 'command takes exactly {} args'.format(self.arg_count))
--> 177         return self.exec_function(*args)

c:\users\g-gre-gre058050\qcodes\qcodes\instrument_drivers\american_magnetics\AMI430.py in <lambda>(units)
    154                            get_cmd='FIELD:UNITS?',
    155                            set_cmd=(lambda units:
--> 156                                     self._update_units(field_units=units)),
    157                            val_mapping={'kilogauss': 0,
    158                                         'tesla': 1})

c:\users\g-gre-gre058050\qcodes\qcodes\instrument_drivers\american_magnetics\AMI430.py in _update_units(self, ramp_rate_units, field_units)
    452         else:
    453             self.write("CONF:FIELD:UNITS {}".format(field_units))
--> 454             field_units = self.field_units.val_mapping[field_units]
    455 
    456         # Map to shortened unit names

KeyError: (0, 'setting x2_field_units to kilogauss')
```

### System
**operating system**
Win10

**qcodes branch**
master

**qcodes commit**
SHA-1: 134ddc1de0d5fddf49f65bb32d1257e5bd53bf06

* Merge pull request #1688 from jenshnielsen/newenv

Update conda env to python 3.7

jenshnielsen, 2019-09-06:
I think this is the issue resolved by https://github.com/QCoDeS/Qcodes/pull/1599 which is just waiting on a test on a real instrument.

simonzihlmann, 2019-09-06:
Yes, it looks exactly like this. Sorry, I didn't find  #1599 before. I think one can then close this issue.

closed 2019-09-06 by simonzihlmann

jenshnielsen, 2019-09-06:
No worries. If you have a chance to do so it would be great if you can test the code from that pr and confirm that it resolves the issue

simonzihlmann, 2019-09-06:
The code from #1599 fixes the above mentioned issue.

jenshnielsen, 2019-09-06:
@simonzihlmann Thanks a lot. I will go ahead and merge :)