microsoft/Qcodes #1709: Filename collision due to case-sensitivity in Keysight folder

spauka, 2019-09-11:
When pulling the qcodes repository on Windows, there is a filename collision between the uppercase and lowercase Keysight folders.

The error message is as follows:
```bash
$ git clone https://github.com/QCoDeS/Qcodes.git
Cloning into 'Qcodes'...
remote: Enumerating objects: 1522, done.
remote: Counting objects: 100% (1522/1522), done.
remote: Compressing objects: 100% (655/655), done.

Receiving objects: 100% (112398/112398), 242.65 MiB | 2.58 MiB/s, done.
Resolving deltas: 100% (87395/87395), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'qcodes/instrument_drivers/Keysight/__init__.py'
  'qcodes/instrument_drivers/keysight/__init__.py'
```

I propose we remove the lowercase keysight folder as it has now been deprecated for over 2 years.

### System
Windows/OSX and other OS's with case insensitive file systems.

astafan8, 2019-09-11:
i'm working on macOS and haven't seen this problem. But i'm all for removing that lowercase directory if it is possible in a clean way.

jenshnielsen, 2019-09-11:
@astafan8 Please see the discussion of the issues in #1710

astafan8, 2019-09-11:
@jenshnielsen thanks, already curiously following it :)

PR #4248 (2022-09-12): Remove lower-case keysight folder
Closes #1709

This fixes the case sensitivity issue that duplicate the Keysight folder with a lower case letter. While the net changes appear to just be the removal of the folder, there are five commits that effectively perform the removal through a sequence of renames that should work for both case sensitive file systems (Linux) and case insensitive systems (Windows). Note that newly cloned repos on Windows will no longer have the warning about a filename collision, but existing repos that update after this change will show an unstaged removal of qcodes/instr [... cut]
closed 2022-09-12 by jenshnielsen