microsoft/Qcodes #1736: Add name parameter to Measurement class

mgunyho, 2019-09-26:
The `Measurement` class has the attribute `name`, which is currently hardcoded to an empty string ([here](https://github.com/QCoDeS/Qcodes/blob/master/qcodes/dataset/measurements.py#L686)). Because of this, doing `with meas.run() as datasaver: ...` will always give the dataset the name `"results"` ([here](https://github.com/QCoDeS/Qcodes/blob/master/qcodes/dataset/measurements.py#L582)). Would it be possible to add a `name` argument to the Measurement constructor? So that the `__repr__` of a dataset could be slightly more informative.

Or does this mess up something with the database? As in https://github.com/QCoDeS/Qcodes/issues/898

mgunyho, 2019-09-27:
If I add `name` to the constructor, I guess it should be the first parameter to be consistent with everything else. But this might break some scripts which assume that the signature is `Measurement(experiment, station)`. In the examples the parameters are always named (which is good) but I see that in my own scripts I've done this. For my scripts it's not a big deal, but do you think the constructor should also accept an Experiment object as the name parameter, with an appropriate deprecation warning? Or is this such an easy fix that it's okay to break old scripts?

astafan8, 2019-09-27:
> I guess it should be the first parameter to be consistent with everything else

yet, but backwards compatibility it more important. So, let's have `name` as a kwarg.

> but do you think the constructor should also accept an Experiment object as the name parameter, with an appropriate deprecation warning

sure, that'd be great. But i do not think there's much value in the effort investment that it prescribes.

-------

another question. currently, it is possible to change the `name` after `Measurement` object is instantiated. Shall we keep this behavior? I think it's fair to keep because it allows to run an existing measurement with all the registered parameters under a different name which might be useful. opinions @QCoDeS/core ?

PR #1741 (2019-10-17): Add name parameter to Measurement class
Fixes #1736.

I added the name as the first kwarg to be consistent with other classes, and also a deprecation warning if the first parameter is instead an Experiment.
closed 2019-10-17 by astafan8