microsoft/Qcodes #1737: [feature request] Make database easier to browse

mgunyho, 2019-09-26:
If I have a lot of runs in the same experiment, I'd like to be able to tag some of them as "successful" or "good" etc. so that I can later quickly skim through all of them to find the relevant ones.

The first thing that comes to mind is to add metadata to the datasets, and then modify the `DataSet.__repr__` to show them. Printing the whole metadata dict might be ugly so maybe just some special keys such as "notes" or "tags" could be added to the repr. The biggest problem with this is that the metadata feature is broken right now (e.g. https://github.com/QCoDeS/Qcodes/issues/1444, https://github.com/QCoDeS/Qcodes/issues/898).

The `load_by_run_spec` function is not sufficient, since run_id and the other parameters it accepts are not that descriptive. Perhaps it could be extended to somehow search in the metadata, but that is a bit of a bigger feature to implement.

There could also be a tutorial/example notebook on how to efficiently browse through old datasets. For full-fledged browsing, I see the [inspectr](https://github.com/data-plottr/plottr) project, which could also be perhaps mentioned in the tutorial. I still think that some minor features to help navigate the data should be built-in, either in the dataset repr or some other way.

I can try to write an example notebook and submit a PR, once I figure out a decent way to browse the datasets. It would be nice to hear some tips on this from people with more experience.

astafan8, 2019-09-27:
> I can try to write an example notebook and submit a PR, once I figure out a decent way to browse the datasets.

that is definitely very welcome! thank you for the initiative!

> I'd like to be able to tag some of them as "successful" or "good" etc.

> add metadata to the datasets

> some special keys such as "notes" or "tags"

all these are valid requests and have been long-awaited. To my current understanding, this is not an easy-to-add feature, it requires a bit of design/thinking beforehand. For example, adding selected metadata to `repr` feels like a hack, and it's very difficult to define what should go there and what shouldn't.

------

> It would be nice to hear some tips on this from people with more experience.

one way would be to implement a set of convenient functions (similar to `load_by_run_spec`) that are just encapsulating sqlite queries to the database. The danger here is that we'll end up re-implementing SQL query language in qcodes-python functions (wasted effort). In order to stay away from the danger zone, it might be worth considering just the most relevant use cases, and then implementing functions which solve exactly them.

-------

> ...to somehow search in the metadata...

this may not be that difficult. for example, a simple suggestion: `load_by_metadata(..., tag: str = None, values: Collection[Union[str, int, float, ...]]=None)`, usage example `load_by_metadata(..., tag='goodness', values={'perfect', 'good'})`, where tag is the tag you're wanting to look for, and values are the values of the tag for selecting. Yes, no search by multiple tags, etc. It might be worth considering what the return values and inputs should be so that one could, for example, `load_by_run_spec` first, and then pass the output to `load_by_metadata` (or better call it then `filter_by_metadata`) for finer filtering. All of these functionalities can be implemented via SQL queries, see `qcodes.dataset.sqlite.*` for examples.

So this was to suggest the approach of quick-to-implement simple functions which solve small immediate problems; as opposed the approach of designing the full metadata-related ecosystem in a nice way which is more time-consuming.

again, PRs are welcome.

mgunyho, 2019-10-04:
How about if instead of using metadata, `DataSet` would have a specific attribute for this purpose, e.g. `comment`. This attribute could be restricted to be only a string (and not a nested object like metadata), so it would be much easier to integrate to the database schema.

For me it still seems that the best way to get an overview of the runs in an experiment is to just do `print(experiment)`, and there it would be nice if such comment appeared in the string constructed in `Experiment.__repr__`.

astafan8, 2021-02-11:
@mgunyho have you tried the interactive experiment widget in the jupyter notebook? read more here https://qcodes.github.io/Qcodes/examples/15_minutes_to_QCoDeS.html#Explore-the-data-using-an-interactive-widget , it might already do what you were looking for :)

mgunyho, 2021-02-19:
At least as a first impression, that looks pretty great! It seems to have most features I was asking for. Is adding a note the same as adding some metadata? Have the metadata issues with the database been fixed?

For now, I think this issue can be closed, I'll open new ones if I come across something specific in the widget.

astafan8, 2021-02-19:
> At least as a first impression, that looks pretty great!

Thank you! the creadit goes to @basnijholt  ;)

> Is adding a note the same as adding some metadata?

yes. the note is added as a metadata with a very specific key.

> Have the metadata issues with the database been fixed?

well, not really :) but if you are aware of them, the feautre can be used. for example, the widget is doing it in a careful way :)

> For now, I think this issue can be closed, I'll open new ones if I come across something specific in the widget.

great! thank you! please open new spwcific ones, indeed, as needed :)

closed 2021-02-19 by astafan8